ros/rosdistro #6878: Jade: Check wiki pages

wjwwood, 2015-01-16:
This is a list of wiki pages that I think we should look at and update for Jade if needed:

- [ ] http://wiki.ros.org/actionlib_tutorials/Tutorials/RunningServerAndClientWithNodes at line 138
- [ ] http://wiki.ros.org/actionlib_tutorials/Tutorials/SimpleActionClient(28)Threaded(29) at line 169
- [ ] http://wiki.ros.org/actionlib_tutorials/Tutorials/SimpleActionServer(28)GoalCallbackMethod(29) at line 22
- [ ] http://wiki.ros.org/catkin at line 3
- [ ] http://wiki.ros.org/class_loader at line 4
- [ ] http://wiki.ros.org/compressed_depth_image_transport at line 29
- [ ] http://wiki.ros.org/compressed_image_transport at line 28
- [ ] http://wiki.ros.org/costmap_2d at line 53
- [ ] http://wiki.ros.org/cv_bridge/Tutorials/UsingCvBridgeToConvertBetweenROSImagesAndOpenCVImages at line 27
- [ ] http://wiki.ros.org/dynamic_reconfigure/Tutorials/SettingUpDynamicReconfigureForANode(28)python(29) at line 28
- [ ] http://wiki.ros.org/gazebo at line 8
- [ ] http://wiki.ros.org/gazebo_ros_pkgs at line 21
- [ ] http://wiki.ros.org/image_proc/common at line 9
- [ ] http://wiki.ros.org/image_proc/cturtle at line 9
- [ ] http://wiki.ros.org/image_proc/diamondback at line 40
- [ ] http://wiki.ros.org/image_proc/electric at line 67
- [ ] http://wiki.ros.org/image_transport at line 112
- [ ] http://wiki.ros.org/image_transport_plugins at line 7
- [ ] http://wiki.ros.org/image_view/diamondback at line 182
- [ ] http://wiki.ros.org/interactive_marker_twist_server at line 10
- [ ] http://wiki.ros.org/mjpeg_server at line 9
- [ ] http://wiki.ros.org/navigation/Tutorials/RobotSetup/TF at line 65
- [ ] http://wiki.ros.org/opencv2 at line 12
- [ ] http://wiki.ros.org/os/OSX/Homebrew/troubleshooting at line 28
- [ ] http://wiki.ros.org/pcl/Tutorials at line 21
- [ ] http://wiki.ros.org/pluginlib/pluginlib_pregroovy at line 3
- [ ] http://wiki.ros.org/pluginlib/Tutorials/Writing(20)and(20)Using(20)a(20)Simple(20)Plugin at line 37
- [ ] http://wiki.ros.org/pluginlib at line 5
- [ ] http://wiki.ros.org/ROS/Tutorials/rosdep at line 39
- [ ] http://wiki.ros.org/ROS/Tutorials/SetupEnvironment at line 9
- [ ] http://wiki.ros.org/rosbash at line 9
- [ ] http://wiki.ros.org/rosgraph_msgs at line 15
- [ ] http://wiki.ros.org/rviz/Tutorials/Librviz(3a20)Incorporating(20)RViz(20)into(20)a(20)Custom(20)GUI at line 26
- [ ] http://wiki.ros.org/rviz/Tutorials/Markers(3a20)Basic(20)Shapes at line 134
- [ ] http://wiki.ros.org/rviz/Tutorials/Plugins(3a20)New(20)Display(20)Type at line 26
- [ ] http://wiki.ros.org/rviz/Tutorials/Plugins(3a20)New(20)Dockable(20)Panel at line 26
- [ ] http://wiki.ros.org/rviz/Tutorials/Plugins(3a20)New(20)Tool(20)Type at line 26
- [ ] http://wiki.ros.org/simulator_gazebo/GazeboConfiguration at line 8
- [ ] http://wiki.ros.org/StackReleaseProcess at line 5
- [ ] http://wiki.ros.org/stage at line 7
- [ ] http://wiki.ros.org/stereo_image_proc/disparity_params at line 1
- [ ] http://wiki.ros.org/StyleGuide at line 439
- [ ] http://wiki.ros.org/teleop_twist_keyboard at line 11
- [ ] http://wiki.ros.org/test_macro at line 3
- [ ] http://wiki.ros.org/tf/Overview/Data(20)Types at line 12
- [ ] http://wiki.ros.org/tf/Tutorials/Debugging(20)tf(20)problems at line 30
- [ ] http://wiki.ros.org/tf/Tutorials/Introduction(20)to(20)tf at line 157
- [ ] http://wiki.ros.org/tf/Tutorials/Using(20)Stamped(20)datatypes(20)with(20)tf(3a3a)MessageFilter at line 28
- [ ] http://wiki.ros.org/tf/Tutorials/Writing(20)a(20)tf(20)broadcaster(2028)C(2b2b29) at line 68
- [ ] http://wiki.ros.org/tf/Tutorials/Writing(20)a(20)tf(20)listener(2028)Python(29) at line 33
- [ ] http://wiki.ros.org/theora_image_transport at line 4
- [ ] http://wiki.ros.org/versionexample at line 8
- [ ] http://wiki.ros.org/visualization at line 3
- [ ] http://wiki.ros.org/WikiMacros at line 16

wjwwood, 2015-01-16:
If you are going to process a page, please check it before you start to we don't clobber each other.

wjwwood, 2015-05-06:
@hsu or @scpeters, can one of you have a look at http://wiki.ros.org/gazebo_ros_pkgs and update it for Jade? It looks like it needs content updates, not just wiki macro touch-ups.

scpeters, 2015-05-06:
@wjwwood We are working on it.

wjwwood, 2015-05-07:
@esteve on http://wiki.ros.org/pluginlib the TOC doesn't show. You may need to move the TOC macro call into the included pages.

esteve, 2015-05-07:
@wjwwood fixed.

wjwwood, 2015-05-07:
Ok, I've touched up or delegated all the pages on this issue.

closed 2015-05-07 by wjwwood