ros/rosdistro #6946: Documentation Audit: catkin

wjwwood, 2015-01-22:
After touching some of our REP's I think we need to check up on our catkin documentation.

I've compiled a list of pages or sections of pages which need attention which some notes. I'll be trying to work through this as I have time.

#### REP-0140: Package.xml version 2

- [ ] http://wiki.ros.org/catkin/conceptual_overview#Dependency_Management

#### REP-0008: Update to Python best practices

- [ ] http://wiki.ros.org/catkin/conceptual_overview#setup.py
 - This one seems to have some incorrect information in it about dependencies, etc...
 - Maybe it should be removed and point to another resource on the `setup.py` with catkin.

wjwwood, 2015-12-09:
Checked off an item here based on https://github.com/ros/catkin/pull/769/files

dirk-thomas, 2020-09-01:
Closing due to age and inactivity.

closed 2020-09-01 by dirk-thomas