ros/rosdistro #7090: Package conflicts on Ubuntu 14.04

rhass-r7, 2015-02-04:
The following packages contain conflicting files, preventing the installation of python3-rosinstall

```
(Reading database ... 64690 files and directories currently installed.)
Preparing to unpack .../python3-catkin-pkg_0.2.6-1_all.deb ...
Unpacking python3-catkin-pkg (0.2.6-1) ...
dpkg: error processing archive /var/cache/apt/archives/python3-catkin-pkg_0.2.6-1_all.deb (--unpack):
 trying to overwrite '/usr/bin/catkin_test_changelog', which is also in package python-catkin-pkg 0.2.6-1
Preparing to unpack .../python3-rospkg_1.0.34-1_all.deb ...
Unpacking python3-rospkg (1.0.34-1) ...
dpkg: error processing archive /var/cache/apt/archives/python3-rospkg_1.0.34-1_all.deb (--unpack):
 trying to overwrite '/usr/bin/rosversion', which is also in package python-rospkg 1.0.34-1
Preparing to unpack .../python3-rosdistro_0.4.0-1_all.deb ...
Unpacking python3-rosdistro (0.4.0-1) ...
dpkg: error processing archive /var/cache/apt/archives/python3-rosdistro_0.4.0-1_all.deb (--unpack):
 trying to overwrite '/usr/bin/rosdistro_reformat', which is also in package python-rosdistro 0.4.0-1
Errors were encountered while processing:
 /var/cache/apt/archives/python3-catkin-pkg_0.2.6-1_all.deb
 /var/cache/apt/archives/python3-rospkg_1.0.34-1_all.deb
 /var/cache/apt/archives/python3-rosdistro_0.4.0-1_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

dirk-thomas, 2015-02-04:
The Python 2 and Python 3 ROS packages are known to be not side-by-side installable. This has been reported for specific packages before (e.g. https://github.com/ros-infrastructure/rospkg/issues/71). There is not obvious way to fix this (since some packages provide global binaries). Therefore the best approach is to keep the Python 3 related stuff in a virtual env.

We also can't express the conflict in the Debian control file until stdeb (which we use for releasing the Python packages) merges and releases our pull request to add that feature: https://github.com/astraw/stdeb/pull/88

rhass-r7, 2015-02-04:
Should we close this issue in favor of re-opening #71 until the upstream PR is merged and released?

dirk-thomas, 2015-02-04:
As soon as stdeb supports the feature we will add the conflict entries to all Python packages. But this won't make them side-by-side installable. They will still conflict with each other - just in a more graceful way.

dirk-thomas, 2015-02-16:
`stdeb` 0.8.4 has just been released (https://github.com/astraw/stdeb/commit/c1c4eac6a28cdd6d37c64e298acdb30bb73572c4). We should release all Python packages (which have global binaries) with their conflicts listed in the stdeb.cfg file:

* [ ] bloom @wjwwood 
* [x] catkin_pkg @dirk-thomas 0.2.7 (https://github.com/ros-infrastructure/catkin_pkg/commit/d82efa91b7da077396db03b3197a0ccbf779d7ce)
* [ ] catkin_tools @wjwwood 
* [ ] rosdep @wjwwood 
* [x] rosdistro @dirk-thomas 0.4.1
* [ ] rosinstall @wjwwood 
* [ ] rosinstall_generator @wjwwood 
* [x] rospkg @dirk-thomas 1.0.35
* [ ] vcstool @dirk-thomas 
* [ ] vcstools @wjwwood 
* [ ] wstool @wjwwood 

Please make sure to update your local environment before releasing new Python packages: https://github.com/ros-infrastructure/ros_release_python/commit/7c4c3c7dd344b069b3250e470c1804dee84a47b0

wjwwood, 2015-02-23:
Does `vcstools` need this? It doesn't install any scripts.

dirk-thomas, 2015-02-23:
If it doesn't it does not need to declare them. It should be perfectly possible to install the Python 2 and 3 packages of it side-by-side then.

wjwwood, 2015-02-23:
I'll try it.

wjwwood, 2015-02-23:
I'm able to install and use it in both Python and Python3 at the same time.

wjwwood, 2015-02-24:
Ok, my packages are done. I'm about to sync to public and tryout the installations on the new versions now.

wjwwood, 2015-02-24:
Ok, I was able to install all of the new versions from `pip` and `apt-get`.

tfoote, 2015-02-27:
all packages checked off, except vcstool which is parallel installable. They correctly conflict now.

dirk-thomas, 2015-03-01:
I am a bit puzzled. You were able to install both `vcstool` Debian packages at the same time?

It does not work for me and I wouldn't expect it to because both try to install the global `vcs` script.

tfoote, 2015-03-01:
Yeah, it definitely conflicts. I must have done vcstools by accident instead.

dirk-thomas, 2015-03-03:
Released vcstool 0.1.8.

closed 2015-03-03 by dirk-thomas