ros/rosdistro #7148: Question: rosdep yaml consistency

jbohren, 2015-02-09:
So I wanted to make it so [rosindex](rosindex.github.io) can **(1)** determine if a package dependency is a rosdep or if it's another ROS package. Additionally, I wanted to **(2)** show a page listing the details of that rosdep (i.e. which platforms it's available on and which distributions). In the future, this could be used to **(3)** automatically declare whether a given package is supported on a given platform.

1. Is possible just by reading the identifiers in the rosdep yaml files, which is fine
2. Creates some issues though, because of the variability in the way rosdep keys are listed. I've found that the following forms exist in the current yaml files:

    ```yaml
#### CASE A ####

ROSDEP-KEY:
  PLATFORM: array-of-deps

#### CASE B ####

ROSDEP-KEY:
  PLATFORM:
    VERSION: array-of-deps

#### CASE C ####

ROSDEP-KEY:
  PLATFORM:
    VERSION:
      MANAGER: array-of-deps

#### CASE D ####

ROSDEP-KEY:
  PLATFORM:
    VERSION:
      MANAGER:
        packages: array-of-deps

#### CASE E... ugh ####

ROSDEP-KEY:
  PLATFORM:
    MANAGER:
      packages: array-of-deps

#### CASE FFFF... ugh ####

ROSDEP-KEY:
  PLATFORM:
    VERSION:
      packages: array-of-deps 
```

    Without knowing the names of the platforms, platform versions, and package managers a priori, case *E* and case *FFFF* are indistinguishable. Are such keys correct, or is one of these cases wrong? I understand that in most cases, the only rosdep keys you care about in a given instance are the ones needed by your OS, in which case you know what platform/version/etc you're looking for.

3. Is further out but a good thing to have on the horizon, I think.

wjwwood, 2015-02-09:
> Without knowing the names of the platforms, platform versions, and package managers a priori, case E and case FFFF are indistinguishable.

That's correct, but rosdep knows this information, otherwise it wouldn't be able to resolve keys using them.

> Are such keys correct, or is one of these cases wrong?

They are correct, one easy way to check is with `rosdep resolve <key> --os=PLATFORM:VERSION`.

`rosdep` knows how to expand the keys, for example if you try to resolve a key which exists but not for the target OS platform/version:

```
% rosdep resolve ipython --os=gentoo:2014

ERROR: No definition of [ipython] for OS [gentoo]

No definition of [ipython] for OS [gentoo]
	rosdep key : ipython
	OS name    : gentoo
	OS version : 2014
	Data: debian:
- ipython
fedora:
- python-ipython
macports:
- py27-ipython
ubuntu:
  lucid:
  - ipython
  maverick:
  - ipython
  natty:
  - ipython
  oneiric:
  - ipython
  precise:
  - ipython
  quantal:
  - ipython
  raring:
  - ipython
  saucy:
  - ipython
  trusty:
  - ipython
  trusty_python3:
  - ipython3
  utopic:
  - ipython
```

Maybe you can look into `rosdep` and see how it does it. I wouldn't recommend trying to interpret the `rosdep` yaml files yourself.

----

Case E is using what I would call an implicit `any_version` key and Case F uses what I would call the implicit `default_installer` key. These are concepts I came up with when I described the new YAML format for `xylem`, which was then greatly improved and expanded on by @NikolausDemmel. Here is the design doc page on the rules files:

http://xylem.readthedocs.org/en/latest/spec.html

And the description of the format in the source code:

https://github.com/catkin/xylem/blob/master/xylem/specs/plugins/rules.py#L15-L244

Some of the design of the new rules format was done to formally describe how this expansion worked and to provide tools to make it just as concise but more explicit where possible.

Anyways, I link you to that stuff not to say "wait for this", but just because it might help you make sense of the current state of `rosdep`'s rules. Also at some point @NikolausDemmel was working on a script to automatically convert the `rosdep` rules into the new `xylem` format, but I can't find that right now, maybe he can point you at it, because it would also be useful for an example of something that interprets the `rosdep` rules directly.

jbohren, 2015-02-09:
Thanks for the info @wjwwood.

> Maybe you can look into rosdep and see how it does it. I wouldn't recommend trying to interpret the rosdep yaml files yourself.

As far as I can tell, it wouldn't be too hard to generate a full table of rosdeps if you know the lists of platforms, versions and their respective package managers. So I'm just going to do that, and then query it for each key.

jbohren, 2015-02-11:
Closing this, I've got a satisfactory solution.

closed 2015-02-11 by jbohren