ros/rosdistro #7151: rosdep: bad rule for python-pymodbus in python.yaml

jbohren, 2015-02-09:
What's going on here (`debian: precise`): https://github.com/ros/rosdistro/blob/master/rosdep/python.yaml#L951

Added in: https://github.com/ros/rosdistro/commit/453b138b5bc1da7a6972dd398a75998d0cb09e05

```yaml
python-pymodbus:
  debian:
    precise: [python-pymodbus]
  fedora: [pymodbus]
  ubuntu:
    precise: [python-pymodbus]
    raring: [python-pymodbus]
    saucy: [python-pymodbus]
    trusty: [python-pymodbus]
```

wjwwood, 2015-02-09:
Looks like a typo, that would never resolve for Debian.

tfoote, 2015-02-09:
Yeah, it's from when paul backfilled a lot of debian. It looks like he was
a little too quick on the copy and paste:
https://github.com/ros/rosdistro/commit/453b138b5bc1da7a6972dd398a75998d0cb09e05#diff-701c2f61a9f7ac8fc393ccc6630c8fa5R202

​

commit 0daa7307 (2015-02-20): Merge pull request #7319 from ros/pymodbus_debian

fixing debian python-pymodbus rule. Fixes #7151
commit 66b2f40f (2015-02-20): fixing debian python-pymodbus rule. Fixes #7151
closed 2015-02-20 by tfoote