ros/rosdistro #7326: removing unsupported versions of a distro

vrabaud, 2015-02-21:
We have quite a few keys that still mention distributions we do not support anymore.
Keeping them often prevents us from having a generic dependency for all versions of a common distro. Thus, whenever a new version of a distro comes out, we have to update the rosdep which is cumbersome.
E.g.:  https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L27 (and I just took the first one).
If you agree with removing all versions of Ubuntu prior to Precise, I'll gladly make a pull request.

dirk-thomas, 2015-02-21:
Some people still build very old versions of ROS from source. I assume removing these "old" distros will make it much more difficult for them to install dependencies.

vrabaud, 2015-02-21:
still, do we want to support versions of Ubuntu that are not even supported by Canonical anymore ? (like karmic or soon lucid)
For the range, that could work, but what will happen once Ubuntu reaches z ? Nobody knows ... So we would have to maintain a list of previous Ubuntu distros ? Seems like added complexity for a very small and undefined number of users no ?

dirk-thomas, 2015-02-21:
A recent download statistic showed that the number of downloads for EOLed ROS distros as well as EOLed Ubuntu distros is not neglectable.

As mentioned before this is only a shortcoming of the tooling. If rosdep would allow to specify (open) ranges of distros we can keep any kind of old entries without "paying" for it with any overhead.

Alessandro-Barbieri, 2017-06-16:
I propose to remove `squeeze` `lucid` `natty` `maverick` as them are history now.

dirk-thomas, 2020-09-01:
Closing due to age and inactivity.

closed 2020-09-01 by dirk-thomas