ros/rosdistro #7346: devel-indigo-openrtm_aist devel job continuously triggering

tfoote, 2015-02-24:
Has been triggering continuously http://jenkins.ros.org/job/devel-indigo-openrtm_aist/

It looks to correctly be a branch. Thus not a reoccurance of #3989  Not sure why it's continuously triggering. 

I've disabled the job for now.

tfoote, 2015-02-24:
Also for hydro: http://jenkins.ros.org/job/devel-hydro-openrtm_aist

tfoote, 2015-02-25:
cc/ @k-okada @130s 
It looks like this is an issue upstream: https://issues.jenkins-ci.org/browse/JENKINS-27115 until that's resolved this won't be resolvable. To that end since this is causing several hundred builds per day on the farm I'm going to recommend that we remove the devel entries for these packages. 

This will result in them not being available from "source" via rosinstall, but since it's in the release repository anyway, it's not actually the source anyway. And this entry will only update when bloomed so it's not subject to the standard development cycles.

k-okada, 2015-02-25:
Ok,removed these entries https://github.com/ros/rosdistro/pull/7354

Please let me know if there are something that we should fix in our side

tfoote, 2015-02-25:
Thanks @k-okada I'm working on finding a work around. It should not require anything from your side. 

When/if we find the work around we should be able to reenable the jobs.

closed 2015-02-25 by tfoote