ros/rosdistro #7484: Migrate ros-dry-releases repo from Google Code

dirk-thomas, 2015-03-12:
Google Code will shut down Jan 2016: http://google-opensource.blogspot.com/2015/03/farewell-to-google-code.html

* We need to migrate our dry release repo somewhere else: https://code.google.com/p/ros-dry-releases/
* Update involved tools to continue to work (eventually at least in read-only mode)
* Update docs to state the new location / workflow

tfoote, 2015-03-12:
I'm trying out sourceforge's importer. It's running now.

tfoote, 2015-03-12:
Project successfully imported here: https://sourceforge.net/p/ros-dry-releases/

Direct http access is still working for subdirectories. For example all the stacks: http://svn.code.sf.net/p/ros-dry-releases/code/download/stacks/

I think we can basically do a search and replace for the google code url and replace with the sf.net one. A [quick search](https://github.com/search?p=1&q=ros-dry-releases&type=Code&utf8=%E2%9C%93) only finds 17 code results including a few forked instances. 

This should work trivially for read only access. With hydro technically EOLd we don't need to support write access.

tfoote, 2016-12-23:
I've opened prs to change the url for the three repos shown in the search. Closing as I don't think we need to track this here anymore.

closed 2016-12-23 by tfoote