ros/rosdistro #8008: Jade: Release Check List

wjwwood, 2015-04-20:
This ticket is used to collect things that need to be done for the upcoming Jade release.

Documentation:

- Update verbiage on these wiki pages, e.g. 'upcoming release' vs. 'current release':
 - [ ] http://wiki.ros.org/jade/Installation

dirk-thomas, 2015-04-20:
Referencing #6435 and #6878.

dirk-thomas, 2015-05-20:
* [x] update REP 142 to move tutorials to desktop metapackage: ros-infrastructure/rep#100
* [ ] update REP 142 to include geneus for Jade
* [ ] release metapackages

wjwwood, 2015-05-22:
@dirk-thomas @tfoote @esteve I'm going to start working on these. I might delegate some tasks. Let me know if you don't have time to do it.

wjwwood, 2015-05-22:
@dirk-thomas can you tell me how to do the "Update symlink for default API doc" item or do it for me?

esteve, 2015-05-22:
K-turtle will be named Kinetic Kame.

wjwwood, 2015-05-22:
@esteve it's supposed to be a secret :stuck_out_tongue:.

tfoote, 2015-05-22:
@wjwwood I created a visualization of the Jade development cycle up to today: https://www.youtube.com/watch?v=-Bpaz94krAo  We've had 40 committers and over 500 packages.

wjwwood, 2015-05-22:
Thanks, I'll include that in my draft email (I'll send it around to you guys tonight and send it tomorrow).

dirk-thomas, 2015-05-22:
@wjwwood  `api` is imply a symbolic link to `indigo/api`. I updated it.

wjwwood, 2015-05-22:
> @wjwwood api is simply a symbolic link to indigo/api. I updated it.

I wasn't sure which server it was on, thanks.

wjwwood, 2015-05-22:
How do I do the www.ros.org updates?

dirk-thomas, 2015-05-22:
I would suggest to put both releases on the install page (http://www.ros.org/install/):
* Indigo (LTS)
* Jade (latest release)

wjwwood, 2015-05-22:
I agree, I've been sort of showcasing them both in each place I can.

wjwwood, 2015-05-23:
I still need to know how to update www.ros.org, has someone done that before? Is that on the internal wiki (I can't seem to access it from here)?

tfoote, 2015-05-23:
@wjwwood I emailed you pointers to the credentials

wjwwood, 2015-05-23:
@tfoote I didn't get an email like that, did you send it to my work account?

tfoote, 2015-05-23:
Yeah, I've resent it with the subject "ros.org credentials"

wjwwood, 2015-05-23:
Ok, I got the www.ros.org updates.

closed 2015-06-25 by wjwwood