ros/rosdistro #8109: Concrete Definitions for ROS variants

cottsay, 2015-04-23:
I would like to include a `comps.xml` file in the RPM repositories that defines RPM groups for each ROS variant. Doing so will integrate with the GUI to the package manager a little more nicely.

I have found the metapackage repository [1] and the various REPs which define them [2], but is there a more clear definition that these packages are generated from? Possibly a YAML file somewhere?

If not, is there any reason that they couldn't be committed to this repo for each distro?

Thanks,

--scott

[1] https://github.com/ros/metapackages
[2] http://www.ros.org/reps/rep-0142.html

dirk-thomas, 2015-04-23:
The metapackages and their dependencies are not generated but manually maintained.

I don't follow what files you want to add but metapackages can not install any additional files within their CMake.

wjwwood, 2015-04-23:
They're completely defined with https://github.com/ros/metapackages. I would recommend you add the installation of the `comps.xml` as a fedora specific patch in the release repository for the `metapackages` repository: https://github.com/ros-gbp/metapackages-release

cottsay, 2015-04-23:
Sorry, let me be more clear.

The `comps.xml` resides with the repository metadata server-side - it isn't packaged for distribution, it is downloaded with the rest of the repo metadata. My goal is to write some python to generate the `comps.xml` programmatically based on the ROS variants, but I don't have any concrete sources for the data besides parsing the `package.xml`s from https://github.com/ros/metapackages.

What would be awesome is to see the YAML examples from http://www.ros.org/reps/rep-0142.html (and others) checked into this repo.

--scott

dirk-thomas, 2015-04-23:
I would rather not create a third location where the variants are listed - it is already in the REP and the package manifest files. why not use the information from the package.xml files? There is a Python library doing the parsing for you and it should take less then a dozen lines of Python code.

wjwwood, 2015-04-23:
I see, I would still just process the `package.xml`'s of the metapackages in the `metapackages` repository. There is (should be) a one-to-one mapping of what is described in REP-0142 and what's in the `package.xml`'s in that repository. Please let us know if it isn't.

If you need to walk dependencies of the `package.xml`'s recursively, you can have a look at my script that tries to estimate what's missing for those variants in Jade which create this wiki page:

http://wiki.ros.org/jade/Planning/Maintenance

The script, along with some others, is located here:

https://github.com/ros-infrastructure/ros.org_scripts/blob/master/wiki.ros.org/jade_metapackage_status.py

cottsay, 2015-04-23:
Thanks, @wjwwood. I'll use that to generate the `comps.xml` then.

I was hoping that this was checked in somewhere and that the `metapackages` repo was generated from more concrete data, but if the `metapackages` repo really is the "source", I'll extract the information from there. It does seem sort-of roundabout, however.

Thank you both!

--scott

closed 2015-04-23 by cottsay