ros/rosdistro #8284: Jade: Community wiki page audit

wjwwood, 2015-05-02:
This is a list of wiki pages which contain at least one instance of the ROS distro wiki seesaw macro.
The line numbers below point to the first instance, not necessarily all instances.

If you look into a wiki page you can comment saying you did and I'll check off the TODO check box. Please try to batch up your comments though to reduce email traffic.

Here is the list of wiki pages:

- [ ] [rosserial_arduino/Tutorials/Time and TF](http://wiki.ros.org/rosserial_arduino/Tutorials/Time and TF)
 - on line 109: `{{{{{{#!wiki version hydro indigo`
- [ ] [win_ros/graveyard/Tutorials/Msvc SDK Projects](http://wiki.ros.org/win_ros/graveyard/Tutorials/Msvc SDK Projects)
 - on line 30: `{{{#!wiki version cturtle`
- [ ] [kobuki_dashboard](http://wiki.ros.org/kobuki_dashboard)
 - on line 28: `{{{#!wiki version groovy`
- [ ] [StyleGuide](http://wiki.ros.org/StyleGuide)
 - on line 439: `{{{{#!wiki version electric fuerte`
- [ ] [Industrial/Install/Legacy](http://wiki.ros.org/Industrial/Install/Legacy)
 - on line 3: `{{{#!wiki version hydro`
- [ ] [rviz/Tutorials/Plugins: New Display Type](http://wiki.ros.org/rviz/Tutorials/Plugins: New Display Type)
 - on line 26: `{{{#!wiki version electric`
- [ ] [mjpeg_server](http://wiki.ros.org/mjpeg_server)
 - on line 9: `{{{#!wiki version groovy`
- [ ] [pr2_arm_navigation_apps](http://wiki.ros.org/pr2_arm_navigation_apps)
 - on line 3: `{{{#!wiki version boxturtle`
- [ ] [Industrial/Industrial_Robot_Driver_Spec](http://wiki.ros.org/Industrial/Industrial_Robot_Driver_Spec)
 - on line 3: `{{{#!wiki version electric`
- [ ] [turtlebot_interactive_markers/Tutorials/UsingTurtlebotInteractiveMarkers](http://wiki.ros.org/turtlebot_interactive_markers/Tutorials/UsingTurtlebotInteractiveMarkers)
 - on line 26: `{{{#!wiki version electric`
- [ ] [moveit_msgs](http://wiki.ros.org/moveit_msgs)
 - on line 8: `{{{#!wiki version groovy`
- [ ] [rosjava](http://wiki.ros.org/rosjava)
 - on line 25: `{{{#!wiki version hydro`
- [ ] [pt_BR/ROS/Tutorials/SetupEnvironment](http://wiki.ros.org/pt_BR/ROS/Tutorials/SetupEnvironment)
 - on line 10: `{{{{{#!wiki version electric fuerte`
- [ ] [collision_environment](http://wiki.ros.org/collision_environment)
 - on line 3: `{{{#!wiki version boxturtle`
- [ ] [ar_sys](http://wiki.ros.org/ar_sys)
 - on line 9: `{{{{#!wiki version indigo`
- [ ] [mit-ros-pkg/KinectDemos/HandDetection](http://wiki.ros.org/mit-ros-pkg/KinectDemos/HandDetection)
 - on line 17: `{{{{#!wiki version cturtle`
- [ ] [ROS/Tutorials/rosdep](http://wiki.ros.org/ROS/Tutorials/rosdep)
 - on line 39: `{{{{{#!wiki version groovy`
- [ ] [rocon_python_comms](http://wiki.ros.org/rocon_python_comms)
 - on line 21: `{{{#!wiki version electric fuerte groovy hydro`
- [ ] [yocs_velocity_smoother](http://wiki.ros.org/yocs_velocity_smoother)
 - on line 5: `{{{#!wiki version groovy`
- [ ] [neural_network](http://wiki.ros.org/neural_network)
 - on line 10: `{{{#!wiki version groovy`
- [ ] [pr2_interactive_manipulation](http://wiki.ros.org/pr2_interactive_manipulation)
 - on line 3: `{{{#!wiki version diamondback`
- [ ] [rosjava_tools](http://wiki.ros.org/rosjava_tools)
 - on line 9: `{{{#!wiki version groovy`
- [ ] [pr2_tabletop_manipulation_apps/Tutorials/Starting the Manipulation Pipeline](http://wiki.ros.org/pr2_tabletop_manipulation_apps/Tutorials/Starting the Manipulation Pipeline)
 - on line 28: `{{{#!wiki version electric`
- [ ] [UWU_Bot](http://wiki.ros.org/UWU_Bot)
 - on line 17: `{{{#!wiki version electric`
- [ ] [abb](http://wiki.ros.org/abb)
 - on line 3: `{{{#!wiki version fuerte`
- [ ] [evart_bridge](http://wiki.ros.org/evart_bridge)
 - on line 3: `{{{#!wiki version electric`
- [ ] [prosilica_gige_sdk](http://wiki.ros.org/prosilica_gige_sdk)
 - on line 8: `{{{#!wiki version boxturtle cturtle diamondback electric fuerte groovy`
- [ ] [turtlebot_msgs](http://wiki.ros.org/turtlebot_msgs)
 - on line 7: `{{{{#!wiki version hydro`
- [ ] [Robots/Husky](http://wiki.ros.org/Robots/Husky)
 - on line 5: `{{{#!wiki version fuerte`
- [ ] [gazebo](http://wiki.ros.org/gazebo)
 - on line 8: `{{{#!wiki version cturtle`
- [ ] [kobuki_auto_docking](http://wiki.ros.org/kobuki_auto_docking)
 - on line 3: `{{{#!wiki version groovy`
- [ ] [husky_description](http://wiki.ros.org/husky_description)
[... 498 lines cut ...]
 - on line 3: `{{{#!wiki version groovy`
- [ ] [rocon_console](http://wiki.ros.org/rocon_console)
 - on line 3: `{{{#!wiki version hydro`
- [ ] [rocon_multimaster/Tutorials](http://wiki.ros.org/rocon_multimaster/Tutorials)
 - on line 5: `{{{#!wiki version electric fuerte`
- [ ] [turtlebot/Tutorials/Looking at Camera Data](http://wiki.ros.org/turtlebot/Tutorials/Looking at Camera Data)
 - on line 3: `{{{#!wiki version electric fuerte`
- [ ] [rosjava_build_tools/Tutorials](http://wiki.ros.org/rosjava_build_tools/Tutorials)
 - on line 3: `{{{#!wiki version cturtle diamondback electric fuerte groovy`
- [ ] [head_pose_estimation](http://wiki.ros.org/head_pose_estimation)
 - on line 8: `{{{#!wiki version groovy`
- [ ] [th/ROS/Tutorials/rosdep](http://wiki.ros.org/th/ROS/Tutorials/rosdep)
 - on line 40: `{{{{{#!wiki version groovy`
- [ ] [pt_BR/ROS/Tutorials/rosdep](http://wiki.ros.org/pt_BR/ROS/Tutorials/rosdep)
 - on line 40: `{{{{{#!wiki version groovy`
- [ ] [rosserial_arduino/Tutorials/Arduino Oscilloscope](http://wiki.ros.org/rosserial_arduino/Tutorials/Arduino Oscilloscope)
 - on line 110: `{{{{{{#!wiki version hydro indigo`
- [ ] [mit-ros-pkg/KinectDemos/ImpromptuButton](http://wiki.ros.org/mit-ros-pkg/KinectDemos/ImpromptuButton)
 - on line 12: `{{{{#!wiki version cturtle`
- [ ] [industrial_extrinsic_cal](http://wiki.ros.org/industrial_extrinsic_cal)
 - on line 3: `{{{#!wiki version hydro`
- [ ] [rosjava/Tutorials](http://wiki.ros.org/rosjava/Tutorials)
 - on line 3: `{{{#!wiki version cturtle diamondback electric fuerte groovy`
- [ ] [actionlib_tutorials/Tutorials/SimpleActionClient(Threaded)](http://wiki.ros.org/actionlib_tutorials/Tutorials/SimpleActionClient(Threaded))
 - on line 169: `{{{{{#!wiki version electric`

trainman419, 2015-05-02:
Teleop_twist_keyboard docs are up to date for jade

gavanderhoorn, 2015-05-08:
I've checked all flagged ROS-Industrial pages. Following packages have not been released into Jade, rockers / seesaws are good:

 - abb
 - adept
 - fanuc
 - fanuc_driver
 - fanuc_experimental
 - industrial_core
 - industrial_extrinsic_cal
 - motoman
 - simple_message
 - staubli
 - staubli_experimental
 - universal_robot
 - ur_driver

Also checked pages. These were updated:
 - [Industrial/Industrial_Robot_Driver_Spec/HardwareCompatibility](http://wiki.ros.org/Industrial/Industrial_Robot_Driver_Spec/HardwareCompatibility)
 - [Industrial/supported_hardware](http://wiki.ros.org/Industrial/supported_hardware)

Deleted:
 - [Industrial/Install/Legacy](http://wiki.ros.org/Industrial/Install/Legacy)

gavanderhoorn, 2015-05-08:
Also checked `view_controller_msgs` and `rviz_animated_view_controller`: have also not yet been released.

vrabaud, 2015-05-08:
@wjwwood : which dump did you use to generate that data ? theora_image_transport was modified in January. And versionedexample does not exist anymore

wjwwood, 2015-05-08:
I got this list quite a while ago, though I merged a more recent list into at some point recently. Some of the pages may already be out of date.

wjwwood, 2015-06-25:
@dirk-thomas I pruned this list using https://github.com/ros/rosdistro/issues/6878

wjwwood, 2017-06-05:
I'm going to close this now as Jade is going EOL. Anyone can continue to use it as a reference to update potentially out-of-date wiki pages.

closed 2017-06-05 by wjwwood