ros/rosdistro #8702: Package `orocos_kdl` is missing in `rosdep/base.yaml`

carlosjoserg, 2015-06-17:
Hi, is there a reason why `orocos_kdl` is in the distribution as a catkin package (at least for indigo [here](https://github.com/ros/rosdistro/blob/master/indigo/distribution.yaml#L5555)), but it does not appear in [rosdep](https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml)?

wjwwood, 2015-06-17:
Anything listed in `rosdep/*.yaml` files are dependencies which are ***not*** released ROS packages. So since it is released through our build farm and has an entry in one or more of the `distribution.yaml` files, it does not need an entry in the rosdep files. You can see it working like this:

```
% rosdep resolve --os ubuntu:trusty --rosdistro indigo orocos_kdl
#apt
ros-indigo-orocos-kdl
```

(If this answers your question, please close the issue. Thanks!)

carlosjoserg, 2015-06-17:
Thanks for the quick reply. That works for me, and if I type `roscd orocos_kdl`, I go to  `/opt/ros/indigo/share/orocos_kdl` where there is `package.xml`.

So, let me reformulate the question...

I see that orocos_kdl is treated as a system dependency in the [kdl_parser cmake file](https://github.com/ros/robot_model/blob/indigo-devel/kdl_parser/CMakeLists.txt#L11), but since it is catkin package, I thought of using it like
``` cmake
find_package(catkin REQUIRED COMPONENTS
  orocos_kdl
)
```
in my package. But that gives an error that I thought was due to the rosdep files, whence the issue.

Could you give me a hint why that does not work or what am I missing?

wjwwood, 2015-06-17:
So, `orocos_kdl` is what we call a "third party, non-catkin package", see:

http://www.ros.org/reps/rep-0136.html

So while it is built on our build farm and installs a `package.xml`, it does not actually use `catkin`. There fore you cannot treat it as a catkin package.

carlosjoserg, 2015-06-17:
Got it. It was the `package.xml` ⇄ catkin-package association that was wrong from my side, but it makes sense after reading that.

Thanks a lot, I'm closing the issue.

closed 2015-06-17 by carlosjoserg