ros/rosdistro #8834: Build server script going mad?

cyborg-x1, 2015-07-01:
I was just checking my traffic on h4r_thermapp_camera and well see yourself:

[image]

As I do not think, 8 cloners  would download it so often (>500 times), I guess there is a script going completely mad, I guess it could be on the build server ... maybe you should check, if there is an error on my repo just tell me and I try to fix it.

Regards,

Christian

wjwwood, 2015-07-01:
There's really no way to know where that's coming from. It could be from http://rosindex.github.io/ for example. I would guess that our farm might generate not more than ~5 clones per release you do, but I don't know that for sure. And we only close the source repo for prerelease, doc, and devel jobs. It could be the devel job, which builds every time you commit, but currently there's no way to turn that off except by removing your `source` entry in the `distribution.yaml` file. In the new build farm (which isn't out yet) you can opt out of devel jobs.

cyborg-x1, 2015-07-01:
Well actually I don't care, I do not pay the traffic. Just wanted to tell somebody, that there might be a problem with a script.

dirk-thomas, 2015-07-01:
You (as the maintainer) should be able to check the build server jobs for your packages. Simply follow these links:

* http://www.ros.org/debbuild/indigo.html?q=h4r_thermapp_camera
* The link on the first square in each column brings you to the jobs:
  * http://jenkins.ros.org/view/Isrc/job/ros-indigo-h4r-thermapp-camera_sourcedeb/
  * http://jenkins.ros.org/job/ros-indigo-h4r-thermapp-camera_binarydeb_saucy_amd64/
  * http://jenkins.ros.org/job/ros-indigo-h4r-thermapp-camera_binarydeb_saucy_i386/
  * http://jenkins.ros.org/job/ros-indigo-h4r-thermapp-camera_binarydeb_trusty_amd64/
  * http://jenkins.ros.org/job/ros-indigo-h4r-thermapp-camera_binarydeb_trusty_i386/

None of them has ran frequently in the past.

If you would create a wiki page for your package (http://wiki.ros.org/h4r_thermapp_camera) it will also contain the links to these jobs (also `devel` and `doc` jobs) if you register your repository for these.

cyborg-x1, 2015-07-01:
So maybe somebody else does that then and his script is going mad, or he is trying to attack github over my repository. And yes I will do all the wiki page stuff later, currently I am working on something else. Actually I plan to release and do documentation for all my packages, but I guess we all know that documentation is often the least important thing for the one writing the software.  -> Time pressure ... ;-)

wjwwood, 2015-07-01:
Thanks for letting us know; I'll close this for now.

closed 2015-07-01 by wjwwood

tfoote, 2015-07-03:
There was a problem with your use of subdirectories. I pulled the source entry 4 days ago: https://github.com/ros/rosdistro/commit/3e82b97f9c963192dfc0a07edeace99dfe194474 

linked here: https://github.com/ros/rosdistro/pull/8549#issuecomment-106433542

cyborg-x1, 2015-07-04:
When I look at my other package above, something get messed up with the development branch. 
I guess it should be the develop branch mentioned at version when I look at all the other source entries.
   source:
      type: git
      url: https://github.com/Hacks4ROS/h4r_thermapp_camera.git
      version: develop

tfoote, 2015-07-04:
The problem was that you're using a submodule with an ssh url. The buildservers do not have ssh for checking things out so it failed to checkout every time.

cyborg-x1, 2015-07-04:
So a submodule with an http URL would probably work?

tfoote, 2015-07-05:
it's much more likely to work. We don't generally recommend using submodules usually we suggest using separate repositories.