ros/rosdistro #8912: Find a way to handle dependencies which differ between ROS versions on OS X

wjwwood, 2015-07-13:
For example, Indigo needs Ogre 1.7 and Jade needs Ogre 1.8/9. These versions of Ogre cannot be installed at the same time and come from different Homebrew formulae. We need some way of representing this difference at the package.xml level which also allows users to select which version gets installed based on something like which Homebrew taps are tapped. See: https://github.com/ros/rosdistro/pull/6334

mikepurvis, 2016-07-25:
I'm now supplying custom rosdep overrides in my install script, in order to get the bottled versions of compiled brewed Python deps:

https://github.com/mikepurvis/ros-install-osx/blob/master/rosdeps.yaml

So it would be trivial to have different files for different distros. The issue for me at this point is not having a stable, reliable formula for an Ogre 1.8 or 1.9 that also works with Gazebo.

dirk-thomas, 2020-09-01:
Closing due to age and inactivity.

closed 2020-09-01 by dirk-thomas