ros/rosdistro #9570: can't find virtual/python-argparse

Alessandro-Barbieri, 2015-09-28:
When I try to install ros on gentoo, it try to emerge virtual/python-argparse, but I can't find it anywhere.
https://github.com/ros/rosdistro/blob/master/rosdep/python.yaml#L99

tfoote, 2015-09-28:
If you could submit a PR with the change it would be greatly appreciated. 

@allenh1 FYI

allenh1, 2015-09-28:
I'll look into it as soon as I can. Thanks @tfoote.

allenh1, 2015-09-28:
Ok, that one is a difference between funtoo and gentoo. I will add the ebuild to ros/ros-overlay.

allenh1, 2015-09-28:
I was under the impression that it wouldn't get flagged on a gentoo install.

allenh1, 2015-09-28:
The ebuild has now been added to the overlay. Please run a `layman -S` to update.
 7e65f21ae80e1c322322fbbc0a53b26fceea60bc

Alessandro-Barbieri, 2015-09-28:
After emerge --regen I got this
> Processing virtual/python-argparse
>  * ERROR: virtual/python-argparse-1000::ros-gentoo failed (depend phase):
>  *   API of python.eclass in EAPI="5-progress" not established
>  * 
>  * Call stack:
>  *                     ebuild.sh, line 584:  Called source '/var/lib/layman/ros-overlay/virtual/python-argparse/python-argparse-1000.ebuild'
>  *   python-argparse-1000.ebuild, line   7:  Called inherit 'python'
>  *                     ebuild.sh, line 280:  Called __qa_source '/usr/portage/eclass/python.eclass'
>  *                     ebuild.sh, line  80:  Called source '/usr/portage/eclass/python.eclass'
>  *                 python.eclass, line  34:  Called die
>  * The specific snippet of code:
>  *      die "API of python.eclass in EAPI=\"${EAPI}\" not established"
>  * 
>  * If you need support, post the output of `emerge --info '=virtual/python-argparse-1000::ros-gentoo'`,
>  * the complete build log and the output of `emerge -pqv '=virtual/python-argparse-1000::ros-gentoo'`.
>  * Working directory: '/usr/lib/python2.7/site-packages'
>  * S: '/var/tmp/portage/virtual/python-argparse-1000/work/python-argparse-1000'
> Error processing virtual/python-argparse-1000, continuing...

allenh1, 2015-09-28:
I have changed the definition of argparse. It was once again linked to the wrong one. I am about to pull request on rosdistro. Sorry for the inconvenience.

PR #9580 (2015-09-28): Fixing definition of argparse for gentoo.
Fixes #9570
closed 2015-09-28 by tfoote