ros/rosdistro #9721: rosdep init gives "Website may be down."

ostueker, 2015-10-27:
I'm trying to install ROS on a Linux Cluster without root access according to http://wiki.ros.org/jade/Installation/Source (using pip).

When I reach to the point of using `rosdep init` I get an error saying that `https://raw.github.com - Website may be down`, but checking with wget gives a certificate validation error. (see below)

Checking with a browser I see that the certificate for raw.github.com does not include github subdomains except www, but that opening the full URL (https://raw.github.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list) directly forwards me to https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list which has a correct SSL certificate.

I believe that you should change the URL scheme to use `https://raw.githubusercontent.com` instead of `https://raw.github.com`.

Is there a way I can do that in my local configuration as well? (I have everything installed under `.local/lib/python2.7/site-packages/`)

Thanks.
O.

```
ostueker@clhead:~ $ ~/.local/bin/rosdep  init -v
ERROR: cannot download default sources list from:
https://raw.github.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
Website may be down.

ostueker@clhead:~ $ wget https://raw.github.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
--16:46:20--  https://raw.github.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
           => `20-default.list'
Resolving raw.github.com... 23.235.44.133
Connecting to raw.github.com|23.235.44.133|:443... connected.
ERROR: Certificate verification error for raw.github.com: unable to get local issuer certificate
ERROR: certificate common name `www.github.com' doesn't match requested host name `raw.github.com'.
To connect to raw.github.com insecurely, use `--no-check-certificate'.
Unable to establish SSL connection.
```

tfoote, 2015-10-27:
rosdep init just fetches that file and writes it into /etc/ros/rosdep/sources.list.d/20-default.list

As a manual workaround you can just fetch the file and write it to that location. 

```
$ cat /etc/ros/rosdep/sources.list.d/20-default.list 
# os-specific listings first
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml osx

# generic
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
yaml https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
gbpdistro https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml fuerte

# newer distributions (Groovy, Hydro, ...) must not be listed anymore, they are being fetched from the rosdistro index.yaml instead
$ ls -l /etc/ros/rosdep/sources.list.d/20-default.list 
-rw-r--r-- 1 root root 588 Jan  6  2015 /etc/ros/rosdep/sources.list.d/20-default.list

```

tfoote, 2015-10-27:
refiled against rosdep https://github.com/ros-infrastructure/rosdep/issues/426

closed 2015-10-27 by tfoote

shehzi-khan, 2017-05-09:
I've installed ROS Kinetic on Linux Mint Sarah (Ubuntu Code name Xenial) using following command.

   ` sudo apt-get install ros-kinetic-desktop-full`

Which executed without any issue. After that I'm trying to execute 

`    sudo rosdep init`

But it gives following error.

```
    ERROR: cannot download default sources list from:
    https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
    Website may be down.
```

However I can access the said link from my browser. The workaround to create write the file manually seems not to work for me, as I can not find /etc/ros in my filesystem.

trDjango, 2017-06-18:
1.5 Initialize rosdep
`sudo rosdep init`
`rosdep update`
But it gives following error.
`ERROR: cannot download default sources list from:
https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
`
In this case what do I do, help..

tfoote, 2017-06-19:
The content is currently available and there's been no indications of issues. You should try again and if you're still having issues check your connectivity for firewalls or proxies that may be limiting access.

trDjango, 2017-06-19:
Thank you for your answer, but I can open the https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list link in my browser, but the same error persists on the command line. `  sudo -E rosdep init` METHODOLOGY ASSEMBLY FAILED AGAIN.
I did the following operations again failed
`sudo apt-get install ca-cacert`
`sudo c_rehash /etc/ssl/certs`
`sudo update-ca-certificates`
I can hopefully finish this installation work, I feel very sorry

tfoote, 2017-06-19:
There are quite a few threads on answers.ros.org related to this: https://www.google.com/search?q=site%3Aanswers.ros.org+cannot+download+default+sources+list It looks like you've already been browsing them. If non of them help you, please open your own question there and show how non of the other solutions work for you so that others might be able to help you out.

Note that testing in a web browser may be traversing a proxy etc.

nusjzx, 2017-08-23:
Is the server down ? 
[image]

tfoote, 2017-08-23:
@nusjzx  This is hosted on github. You can see their status here: https://status.github.com/ It does not look like there's a problem on the server.

It looks like you're behind a varnish proxy that might be having issues. Please ask questions on https://answers.ros.org this is a closed issue and won't get much attention.

shoebahmedadeel, 2018-01-05:
Hi All,
I had been facing the same issue of 
ERROR: cannot download default sources list from:
https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
Website may be down.

This error is because of your system date & time settings. I corrected my systems date and time settings and rosdep init worked.

karimchamaa, 2018-03-01:
I got the same error and fixed it by updating and upgrading the system

kaustubhsridhar, 2018-06-09:
@pmordel thank you so much. that worked!

nand22, 2018-08-12:
I had the same problem. However, disabling the firewall solved the problem for me.

beihaifusang, 2018-09-12:
probably should check if any other progess is heavily using the network, e.g. watching an online movie at the same time. For me, the "git clone" squeeze my ROS out from the internet.

likehuaer, 2019-03-22:
> To solve the error I installed 'ca-certificates' package
> 
> `sudo apt-get install ca-certificates`
> 
> Then it worked for me

also works in my vm linux.

leey0410, 2019-06-15:
> 为了解决错误，我安装了'ca-certificates'包
> 
> `sudo apt-get install ca-certificates`
> 
> 然后它对我有用

it works on my Ubuntu 16.04 LTS

Mansonziqian, 2019-07-28:
> To solve the error I installed 'ca-certificates' package
> 
> `sudo apt-get install ca-certificates`
> 
> Then it worked for me

thank you! that worked!

fjp, 2019-10-16:
To fix this I had to add proxy settings to `/etc/environment`: 

```
http_proxy=http://[user:password@]proxy_server:port/
https_proxy=https://[user:password@]proxy_server:port/
```

The sollution is from https://answers.ros.org/question/54150/rosdep-initialization-error/

AlwaysSoYoung, 2019-10-28:
> To solve the error I installed 'ca-certificates' package
> 
> `sudo apt-get install ca-certificates`
> 
> Then it worked for me

thks, it works for me!

LarryDong, 2020-02-17:
For me, update and upgrade works.

houjiawei11, 2020-04-02:
Using 
```
sudo rosdep update
```
works.

ibrahimqazi, 2020-06-28:
I have tried the above mentioned methods but no success :(

Winnie627, 2020-12-11:
#open hosts file:
sudo gedit /etc/hosts
#add this line:
151.101.84.133  raw.githubusercontent.com
#save and close,then run:
sudo rosdep init
rosdep update

cocodmdr, 2020-12-11:
Does not work for me sadly on Ubuntu Mate 20.04

Winnie627, 2020-12-11:
> > #open hosts file:
> > sudo gedit /etc/hosts
> > #add this line:
> > 151.101.84.133 raw.githubusercontent.com
> > #save and close,then run:
> > sudo rosdep init
> > rosdep update
> 
> Does not work for me sadly on Ubuntu Mate 20.04

Oh it work for me on Ubuntu 18.04

Ericwonne, 2020-12-22:
@pmordel 's solution worked for me as well, solve my problem right away.

nuomifan, 2022-07-02:
> 
I have tried all solutions mentioned above. They don't work.
This works for me. Thank you.

YunTianming675, 2022-09-22:
> I had the same problem. However, disabling the firewall solved the problem for me.

thank you!it worked!

phinchey, 2022-12-29:
> #open hosts file: sudo gedit /etc/hosts #add this line: 151.101.84.133 raw.githubusercontent.com #save and close,then run: sudo rosdep init rosdep update

+1 to this solution. `sudo apt update && sudo apt upgrade` didn't work for me, nor did `sudo apt-get install ca-certificates`. But the /etc/hosts method did!

ychen441, 2023-06-15:
>> sudo -E rosdep init
This works me while all solutions above didn't.

fireworker01, 2023-07-05:
> 

This works on my 18.04. Thanks!!

ObsidianMaximus, 2023-10-05:
> #open hosts file: sudo gedit /etc/hosts #add this line: 151.101.84.133 raw.githubusercontent.com #save and close,then run: sudo rosdep init rosdep update

Yeah this thing worked.

iGuoBao, 2023-10-16:
> #打开主机文件： sudo gedit /etc/hosts #添加这一行： 151.101.84.133 raw.githubusercontent.com #保存并关闭，然后运行： sudo rosdep init rosdep update

thanks!

works on 18.06

Agnessneha, 2023-10-19:
> #open hosts file: sudo gedit /etc/hosts #add this line: 151.101.84.133 raw.githubusercontent.com #save and close,then run: sudo rosdep init rosdep update

worked for me ubuntu server 20.04

yanglh14, 2023-12-15:
> #open hosts file: sudo gedit /etc/hosts #add this line: 151.101.84.133 raw.githubusercontent.com #save and close,then run: sudo rosdep init rosdep update

Worked in 20.04

feryZzza, 2024-01-16:
> #open hosts file: sudo gedit /etc/hosts #add this line: 151.101.84.133 raw.githubusercontent.com #save and close,then run: sudo rosdep init rosdep update

This works on my 22.04. Thanks!

ckiiiik, 2024-03-14:
> > 为了解决错误，我安装了'ca-certificates'包
> > `sudo apt-get install ca-certificates`
> > 然后它对我有用
> 
> it works on my Ubuntu 16.04 LTS

however 20.04 still keep problems

wangyendt, 2024-04-01:
> #open hosts file: sudo gedit /etc/hosts #add this line: 151.101.84.133 raw.githubusercontent.com #save and close,then run: sudo rosdep init rosdep update

works for me, thanks!

wangyendt, 2024-10-19:
This is an auto-sent e-mail. 
 
I am sorry that currently I could not reply your e-mail. I will reply you as soon as possible.