ros/rosdistro #9761: Source Install Fails with "No package vtk-phython available"

danthony06, 2015-11-03:
I am trying to do a source install of Jade onto Fedora 22, using the instructions here: http://wiki.ros.org/jade/Installation/Source. I am trying the desktop-full install. Everything works until I try to resolve the dependencies with the command:

``rosdep install --from-paths src --ignore-src --rosdistro jade -y``

I then get an error that "No package vtk-phython available". I assume this should be vtk-python? It looks like the mistake is here: 

https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L2927-L2929

The full console output is below.

``rosdep install --from-paths src --ignore-src --rosdistro jade -y``

executing command [sudo -H dnf --assumeyes install collada-dom-devel python-matplotlib PyQwt-devel gtk2-devel tinyxml-devel PyQt4 yaml-cpp-devel python-nose pcl-devel opencv-devel ogre-devel libuuid-devel console-bridge-devel sbcl tango-icon-theme cmake cppunit-devel libcurl-devel bzip2-devel pcl-tools python-devel mesa-libGL-devel urdfdom-devel apr-devel vtk-phython vtk-java pcl wxPython-devel qhull-devel numpy urdfdom-headers-devel python-mock boost-devel PyOpenGL log4cxx-devel python-empy PyQt4-devel libtool-ltdl-devel libtheora-devel assimp libtool hddtemp gtest-devel libjpeg-turbo-devel graphviz fltk-devel poco-devel gazebo-devel eigen3-devel python-psutil libogg-devel qt-devel pydot python-pygraphviz lz4-devel python-pillow-qt python-netifaces opencv-python sip-devel mesa-libGLU-devel assimp-devel]

Last metadata expiration check performed 0:02:40 ago on Tue Nov 3 14:10:22 2015.

No package vtk-phython available.

Error: Unable to find a match.

ERROR: the following rosdeps failed to install

dnf: command [sudo -H dnf --assumeyes install collada-dom-devel python-matplotlib PyQwt-devel gtk2-devel tinyxml-devel PyQt4 yaml-cpp-devel python-nose pcl-devel opencv-devel ogre-devel libuuid-devel console-bridge-devel sbcl tango-icon-theme cmake cppunit-devel libcurl-devel bzip2-devel pcl-tools python-devel mesa-libGL-devel urdfdom-devel apr-devel vtk-phython vtk-java pcl wxPython-devel qhull-devel numpy urdfdom-headers-devel python-mock boost-devel PyOpenGL log4cxx-devel python-empy PyQt4-devel libtool-ltdl-devel libtheora-devel assimp libtool hddtemp gtest-devel libjpeg-turbo-devel graphviz fltk-devel poco-devel gazebo-devel eigen3-devel python-psutil libogg-devel qt-devel pydot python-pygraphviz lz4-devel python-pillow-qt python-netifaces opencv-python sip-devel mesa-libGLU-devel assimp-devel] failed

wjwwood, 2015-11-03:
Looks like the rule is wrong, as you pointed out.

danthony06, 2015-11-04:
Fixed with this pull.

https://github.com/ros/rosdistro/pull/9762

closed 2015-11-04 by danthony06