ros/rosdistro #9769: ca-certificates missing from buildfarm slaves

mikepurvis, 2015-11-06:
Curiously, only for Jade builds— Indigo does fine. Example failure:

http://jenkins.ros.org/job/ros-jade-pointgrey-camera-driver_binarydeb_vivid_i386/199/console

The pointgrey_camera_driver package is a bit of a special flower— due to distribution restrictions on the vendor's SDK, it is downloaded on each build, hence the need for certificates to be present.

dirk-thomas, 2015-11-06:
* It looks like that all Indigo jobs work as well as Jade for Trusty and Utopic. Only the Jade Vivid builds are failing.
* The download seems to be done from www.ptgrey.com. I am pretty sure that the build slaves have no specific certificates for that.
* It could be that Ubuntu Vivid comes with a different set of default certificates or that Python 2.7.9 (Vivid) is more strict than 2.7.5 (Trusty)?
* I only see successful builds from before Oct 14th and the history of the failing jobs ends after that date. Did the vivid jobs failed before too? If not something could have changed with the server certificate.

I would suggest to try to run the Python script for downloading the resource on a Vivid machine. Eventually it needs to be modified to work on a Vivid machine.

* Another comment regarding the package: you should update the CMake code to actually fail at CMake time if the expected download doesn't succeed (instead of continuing with the build at fail with a compiler error).
* Curious question: why download the files in the first place if all information to get the data is stored in a public repository??

mikepurvis, 2015-11-06:
> Curious question: why download the files in the first place if all information to get the data is stored in a public repository??

We only redistribute the .so— the headers are used to build but not distributed, per the license.

tfoote, 2015-11-06:
I've been under the impression that ca-certificates was previously in our bootstrap but now is not. And I computed the rdepends on ca-certificates for trusty and vivid and the only rdepends removed was libruby1.9.1 [See this gist](https://gist.github.com/tfoote/898fd73fc61657794f6b) for the full output and diff. 

Along similar lines I cannot find ca-certificates being installed in the successful builds either. Such as: http://jenkins.ros.org/job/ros-jade-pointgrey-camera-driver_binarydeb_trusty_i386/13/consoleText

mikepurvis, 2015-11-06:
[This looks relevant](https://www.cvedetails.com/cve/CVE-2014-9365/):

    The HTTP clients in the (1) httplib, (2) urllib, (3) urllib2, and (4) xmlrpclib libraries in CPython (aka Python) 2.x before 2.7.9 and 3.x before 3.4.3, when accessing an HTTPS URL, do not (a) check the certificate against a trust store or verify that the server hostname matches a domain name in the subject's (b) Common Name or (c) subjectAltName field of the X.509 certificate, which allows man-in-the-middle attackers to spoof SSL servers via an arbitrary valid certificate.

Short of installing the ca-certificates package, it looks like the other option would be [shipping certs with the source](https://docs.python.org/2/library/ssl.html#ssl.SSLContext.load_verify_locations), or simply [turning of cert checking](verify_mode).

mikepurvis, 2015-11-06:
As for the larger "why" of all this. Obviously this behaviour is not really in the spirit of the redistribution terms of the license, since as you say I am distributing a means to trivially acquire the files in question. But I still feel more comfortable from a legal standpoint with distributing the means than distributing the files themselves, since the latter is very clearly forbidden.

And anyway, at least this way, I'm giving them an easy way to turn off the tap— just disable the account whose credentials are baked into the download script.

tfoote, 2015-11-06:
@mikepurvis I agree with the distinction.

Well that's quite a failing of the libraries. I guess the assumptions of security via https were invalid. From here we have a few options. We could disable checking or we can install the certs by some means. I'd suggest that we not disable the certs, as that's just a bad vulnrability. And since it's only a small subset of packages which need the certs I think it might make sense to require that if you're fetching https urls you need to install the certs. I'll note that `wget`, `curl`, and `python*-urllib3` all have dependencies on `ca-certificates` already so it's only a problem if we're fetching from `python-urllib` or `python-urllib2`. 

We've also run into this with [`download_checkmd5.py`](https://github.com/ros/catkin/blob/3b8a3b9607adff98ed32cfd3b6d6ea4aed647424/cmake/test/download_checkmd5.py) in catkin so we should probably add it there too, which will mask any of these packages.

mikepurvis, 2015-11-06:
I was just going to disable the check, but it sounds like a `<build_depend>` on curl would fix the problem by pulling in the dependency transitively.

mikepurvis, 2015-11-09:
Please merge https://github.com/ros/rosdistro/pull/9776 and see if that builds correctly.

(The devel job won't tell us anything as it builds on Trusty)

mikepurvis, 2015-11-09:
Hmm, [still getting the problem](http://jenkins.ros.org/job/ros-jade-pointgrey-camera-driver_binarydeb_vivid_i386/203/console). Perhaps I misunderstood the required fix, since:

 - libcurl3 is being installed at build time, but ca-certificates isn't, so it must've already been installed.
 - the urllib2 request is still failing, so obviously the presence of the certs on the system isn't enough; they have to explicitly mixed in.

A solution [more like this](http://stackoverflow.com/a/14320202), maybe?

tfoote, 2015-11-25:
Yeah, that's unfortunate. I would suggest trying injecting the correct https handler in the script and see if that works. If it does we should consider doing that for the download_upack scripts too.

tfoote, 2016-12-23:
@mikepurvis is this still an issue?

mikepurvis, 2016-12-23:
It appears not: http://build.ros.org/job/Jbin_uV64__pointgrey_camera_driver__ubuntu_vivid_amd64__binary/

closed 2016-12-23 by mikepurvis