ros/rosdistro #9944: Kinetic: Release Check List

esteve, 2015-12-03:
This ticket is used to collect things that need to be done for the upcoming Jade release.

Existing Task Sets:

- [ ] Kinetic kickoff list
- [ ] Kinetic documentation audit
- [ ] Kinetic community documentation audit

jacquelinekay, 2016-05-23:
@dirk-thomas do you know what "remove EOL'd from header doc links" refers to?

dirk-thomas, 2016-05-23:
A wiki page with the package template has links at the top to switch to the documentation of a specific ROS distribution. EOLed distros should be removed from that list in order to not let it grow unbounded.

jacquelinekay, 2016-05-23:
ok, that's what I thought. I'm in favor of this change, but it seems to be a new policy since the templates currently go all the way back to electric.

I think ideally we would hide EOL'd distros on that bar by default and have an option to show the names of EOL'd distros.

wjwwood, 2016-05-23:
ROSBrowse changes: https://github.com/ros-infrastructure/rosbrowse/pull/4

tfoote, 2016-12-23:
Kinetic is out. There's always room to improve documentation.

closed 2016-12-23 by tfoote