ros/rosdistro #9945: Kinetic Kick-off Checklist

esteve, 2015-12-03:
This is a checklist of things to do for the Kinetic kick-off (pre development and development):

- [x] Choose distribution name: Kinetic
- [ ] Add to REP3 with proposed platforms and minimum requirements @esteve https://github.com/ros-infrastructure/rep/pull/106
- [ ] List on ROS wiki with expected release date (http://wiki.ros.org/Distributions)
- [ ] EOL distribution N-3 (Hydro)
- [ ] Add to reprepro-updater list and run script to populate @tfoote W+X supported

mikepurvis, 2016-02-04:
In advance of the full Kinetic debs being ready, it's unfortunate that the independent packages like python-rosdep, python-rosinstall-generator, etc are not available for Wily and Xenial, so that users can get ahead of the game with testing Indigo and Jade packages on the newer versions of system deps.

esteve, 2016-02-06:
bloom: https://github.com/ros-infrastructure/bloom/pull/379
catkin_pkg: https://github.com/ros-infrastructure/catkin_pkg/pull/135
python-rosdep: https://github.com/ros-infrastructure/rosdep/pull/434
rosinstall_generator: https://github.com/ros-infrastructure/rosinstall_generator/pull/32
rospkg: https://github.com/ros-infrastructure/rospkg/pull/96

@mikepurvis sorry for the delay. Is there any other package that needs to be updated? Thanks.

dirk-thomas, 2016-02-06:
@esteve Please look at the checklist from the previous releases for additional Python packages: e.g. #6435.

dirk-thomas, 2016-03-09:
I have released: `catkin`, `genmsg`, `gencpp`, `genlisp`, `genpy`, `roscpp_core`, `message_runtime`, `rosbag_migration_rule`, and `rospack`. Currently I need to wait for a `geneus` before I can proceed.

jacquelinekay, 2016-03-10:
@k-okada, any chance you could make a Kinetic release of `geneus` so that we can proceed with Kinetic core package releases? Many thanks.

jacquelinekay, 2016-03-10:
my bad, I wasn't watching rosdistro and just noticed that Kei has already released geneus...

k-okada, 2016-03-10:
as for Collada, we put deb for wily at
https://launchpad.net/~openrave/+archive/ubuntu/release/+sourcepub/6188757/+listing-archive-extra
https://launchpad.net/~openrave/+archive/ubuntu/release/+sourcepub/6188756/+listing-archive-extra

it seems collada becomes default ubuntu package from xenial, so I think
we'll use default(ubuntu) version of collada for xenial and launch pad
version (from ros repository) for wily, it seems tricky, but better than
install two version of collada.
http://packages.ubuntu.com/xenial/libcollada-dom2.4-dp-dev

wjwwood, 2016-03-10:
I contacted Rosen about updating the openrave ppa, but if you guys already did it then I'll open a or later tonight to add it to our import rules.

k-okada, 2016-03-10:
I haven't talked to Rosen directly, but pretty sure that he's super busy
 and updated the deb for him.
Did you have any response from Rosen? If not, please go ahead.

wjwwood, 2016-03-10:
No nothing yet. I'll set it up.

dirk-thomas, 2016-03-13:
I have released: `message_generation`, `std_msgs`, `ros_comm_msgs`, `ros`, `roslisp`, `ros_comm`, `rosconsole_bridge`, and `common_msgs`.

In order to release `ros_tutorials` we need the Kinetic `turtlesim` icon and I will convert `turtlesim` to Qt5. But that should not block the release of most other downstream packages.

dirk-thomas, 2016-03-16:
I have released: message_generation, std_msgs, ros_comm_msgs, ros, roslisp, ros_comm, rosconsole_bridge, and common_msgs.

In order to release ros_tutorials we need the Kinetic turtlesim icon and I will convert turtlesim to Qt5. But that should not block the release of most other downstream packages.

Update: I have created a kinetic-devel branch for ros_tutorials which uses Qt5.

closed 2016-05-23 by jacquelinekay