ros/rosdistro #10013: Setup daily ROS 1.0/cygwin build in Jenkins

j-rivero, 2015-12-09:
The goal is to automate the process of building the whole set of packages on cygwin from scratch using jenkins and finally produce the binary installer.

The [jenkins job](http://build.osrfoundation.org/job/roscygwin-ci_daily-cygwin64/) is in place and running from our Windows machine at build.osrfoundation.org. The [script](https://bitbucket.org/osrf/release-tools/src/default/jenkins-scripts/cygwin/_ros1_roscygwin_compilation.bash?at=default&fileviewer=file-view-default) is also available.

j-rivero, 2015-12-10:
I'm getting an [error in Jenkins](http://build.osrfoundation.org/job/roscygwin-ci_daily-cygwin64/37/console) when trying to build qt. Looks like [Qt does not support cygwin](http://doc.qt.io/QtSupportedPlatforms/index.html) so probably upstream uses some kind of patching to get built.

dirk-thomas, 2017-04-03:
@j-rivero Is this still something you are pursuing or should this be closed?

j-rivero, 2017-04-03:
> @j-rivero Is this still something you are pursuing or should this be closed?

We can close for now, long time without seeing interest on this. Feel free to reopen if anybody still have interest on it.

closed 2017-04-03 by j-rivero