ros/rosdistro #10331: rosdep libx52pro0 is not installable

tfoote, 2016-01-21:
We ran into this here: http://build.ros.org:8080/view/Idev/job/Idev__h4r_x52_joyext__ubuntu_trusty_amd64/lastFailedBuild/console

However it appears to be a regression in the package. I cannot install it in a clean docker image on ubuntu:trusty. https://gist.github.com/tfoote/dd7e2bbf213765c3f4bb

@cyborg-x1 FYI

Note it seems to have a proposed fix upstream this month: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=767675

cyborg-x1, 2016-01-22:
I saw that, I think of going back to the state where I included the library in my package, then it also works with other Linux distributions where that package is not available from the repository. What do you think about that?

tfoote, 2016-01-22:
It's a bit of a pain, but if the upstream is not available/poorly maintained it's going to give a better user experience. 

The best way to do it is in a way that you embed it in your package but hide the fact from downstream such that in the future you can strip it out and rely on the system installed one and it won't effect anyone else.

tfoote, 2016-01-27:
I'd propose we plan to remove this until it's released and installable.

cyborg-x1, 2016-01-27:
Yes ok, I am currently busy with something else. I will get to it when I am done with the other thing.

closed 2016-03-01 by wjwwood