ros/rosdistro #10370: urllib Error with github

bmagyar, 2016-01-29:
I'm trying to make a prerelease but keep getting an error that seems to be related to SSL of github.
I tried running bloom-release too, to see how that reacts, and the result is the same.
```
$ generate_prerelease_script.py` https://raw.githubusercontent.com/ros-infrastructure/ros_buildfarm_config/master/index.yaml indigo default ubuntu trusty amd64   teleop_tools   --level 1   --output-dir ./
Fetching buildfarm configuration...
Traceback (most recent call last):
  File "/usr/bin/generate_prerelease_script.py", line 281, in <module>
    sys.exit(main())
  File "/usr/bin/generate_prerelease_script.py", line 77, in main
    config = get_config_index(args.config_url)
  File "/usr/lib/python2.7/dist-packages/ros_buildfarm/config/__init__.py", line 51, in get_index
    yaml_str = load_url(url)
  File "/usr/lib/python2.7/dist-packages/ros_buildfarm/config/loader.py", line 63, in load_url
    raise URLError(str(e) + ' (%s)' % url)
urllib2.URLError: <urlopen error <urlopen error [Errno 1] _ssl.c:510: error:140773E8:SSL routines:SSL23_GET_SERVER_HELLO:reason(1000)> (https://raw.githubusercontent.com/ros-infrastructure/ros_buildfarm_config/master/index.yaml)
```
But at the same time I'm able to view the file with Firefox and downloading it with wget works.

All my packages are up to date & tried on two computers (both 64bit, Ubuntu 14.04).

I fired up an ipython and tried urlopen with urllib, urllib2 and even urllib3 with no success.

Digging more into the the source of the error I found this explanation to `SSL23_GET_SERVER_HELLO:reason(1000)`:
```I understand the error message is due to an SSL version error. Using curl shows the same error and more detail - that our client only supports SSLv23 and the server only supports SSLv3:``` (from [here](http://redmine.dguteam.org.uk/issues/1439)

Using urlllib3 I could play with SSL Protocol setups too, but those failed for version reasons.
```
connection_pool = urllib3.PoolManager(ssl_version=ssl.PROTOCOL_SSLv3)
resp = connection_pool.request('GET',url )
...blarg...
SSLError: [Errno 1] _ssl.c:510: error:1408F10B:SSL routines:SSL3_GET_RECORD:wrong version number
```

`curl -v https://raw.githubusercontent.com/ros-infrastructure/ros_buildfarm_config/master/index.yaml` on the other hand works fine doing SSLv3 and (?) TLS handshakes, etc.

Am I the only one with this issue?

tfoote, 2016-01-29:
I cannot reproduce this on my trusty machine using everything from debs. Can you make sure that your running up to date versions of urllib?

bmagyar, 2016-01-29:
I have update everything I could get my hands on... 

Good news though that it's not a general thing.

bmagyar, 2016-01-30:
Ok... So running urllib2.urlopen() with the problematic file above in ipython now on my Macbook with OSX (not Ubuntu) succeeds. Very weird. This means that both laptops I used Ubuntu on are flawed the same way. Thanks for your note anyways, I will close this issue but once I have the solution I'll post it here if relevant.

closed 2016-01-30 by bmagyar