ros/rosdistro #10490: jade distro not installing properly (ros-jade-desktop-full)

dcconner, 2016-02-12:
See http://answers.ros.org/question/226387/ros-jade-desktop-full-not-setting-up-gazebo-properly/

Doing a fresh (after remove and delete) install of ros-jade-desktop-full is not creating the setup scripts in /opt/ros/jade as it has in past.

It is not setting up /usr/share/gazebo scripts folder, and created a /usr/share/gazebo-5.0 folder instead of 5.1 as it did last week.

dirk-thomas, 2016-02-12:
You need to provide more information to allow people to understand what exactly you did and what the problem is.

From the limited description I don't know which Ubuntu distribution you are using. Assuming that you use *Trusty* and have installed the package `ros-jade-catkin` (which is part of `ros-jade-desktop-fill`) from `packages.ros.org` I checked the available Debian package: http://packages.ros.org/ros/ubuntu/pool/main/r/ros-jade-catkin/ros-jade-catkin_0.6.16-0trusty-20160110-082557-0800_amd64.deb
The unpacked package does contain the file `/opt/ros/jade/setup.*` therefore I don't think it is a systematic problem but something in your steps.

Also we are running hundreds of CI jobs for Jade packages and all of them would fail if Jade would not come with these setup files.

gavanderhoorn, 2016-02-12:
Related ROS Answers question (confusingly): [ros-jade-desktop-full not setting up gazebo properly](http://answers.ros.org/question/226387/ros-jade-desktop-full-not-setting-up-gazebo-properly/).

dcconner, 2016-02-12:
Ubuntu 14.04
I started getting odd issue with the gazebo setup (the above ROS Answers) question, and then when I tried re-installing I got the issue with  ROS setup.

It might be something odd with this new computer, but I am not sure what it could be.  

I've done this quite a bit in the past without issue, so I thought it might be related to some recent change in the ROS deb, so I decided to post the ticket.

If your CI jobs are testing fresh installs, then it is likely something on my end.  

I'll keep looking, and update this if I find the issue.

tfoote, 2016-02-12:
I've just installed jade desktop full on a blank machine it installs the setup files for jade successfully and the gazebo packages appear to be in order. 

Here's the full log. 

Please keep iterating on the answers question with more specifics of how to reproduce your problem.

As this does not appear to be a general problem at the level of the rosdistro, I'm going to close this ticket.

closed 2016-02-12 by tfoote

dcconner, 2016-02-12:
OK.  Thanks for checking.  I updated the question on ROS answers question with additional details, including a screen shot of the immediate output of running install ros-jade-desktop-full

I suppose there is something whacked with my Ubuntu setup, but I have know clue what it could be.

dcconner, 2016-02-13:
I did a apt-get clean, and reinstalled and not I get the ros setup files.

I do NOT get the /usr/share/gazebo/setup.sh that is normally there.   I notice you are getting Gazebo 5.0 as I am.  My installations from a few weeks ago was getting Gazebo 5.1

If I try to run the gazebo using roslaunch gazebo_ros empty_world.launch I get an error complaining about the missing setup and gz* files

I tried to manually add the setup from another computer and change to 5.0 and get
'''
setting /run_id to ef838404-d1eb-11e5-9421-b8ee65b7a4b4
process[rosout-1]: started with pid [4534]
started core service [/rosout]
process[gazebo-2]: started with pid [4537]
process[gazebo_gui-3]: started with pid [4545]
/opt/ros/jade/lib/gazebo_ros/gzclient: 17: /opt/ros/jade/lib/gazebo_ros/gzclient: gzclient: not found
/opt/ros/jade/lib/gazebo_ros/gzserver: 22: /opt/ros/jade/lib/gazebo_ros/gzserver: gzserver: not found
[gazebo-2] process has died [pid 4537, exit code 127, cmd /opt/ros/jade/lib/gazebo_ros/gzserver -e ode worlds/empty.world __name:=gazebo __log:=/home/pcse_admin/.ros/log/ef838404-d1eb-11e5-9421-b8ee65b7a4b4/gazebo-2.log].
log file: /home/pcse_admin/.ros/log/ef838404-d1eb-11e5-9421-b8ee65b7a4b4/gazebo-2*.log
[gazebo_gui-3] process has died [pid 4545, exit code 127, cmd /opt/ros/jade/lib/gazebo_ros/gzclient __name:=gazebo_gui __log:=/home/pcse_admin/.ros/log/ef838404-d1eb-11e5-9421-b8ee65b7a4b4/gazebo_gui-3.log].
log file: /home/pcse_admin/.ros/log/ef838404-d1eb-11e5-9421-b8ee65b7a4b4/gazebo_gui-3*.log
'''

dcconner, 2016-02-13:
Looking the gz files in /opt/ros/jade/lib/gazebo_ros, 
'''
pcse_admin@RoboticsLaptop:~$ cd /opt/ros/jade/lib/gazebo_ros/
pcse_admin@RoboticsLaptop:/opt/ros/jade/lib/gazebo_ros$ ls
debug  gazebo  gdbrun  gzclient  gzserver  perf  spawn_model
pcse_admin@RoboticsLaptop:/opt/ros/jade/lib/gazebo_ros$ ls -altr
total 56
-rwxr-xr-x  1 root root 13041 Aug 15 20:32 spawn_model
-rwxr-xr-x  1 root root   572 Aug 15 20:32 perf
-rwxr-xr-x  1 root root   657 Aug 15 20:32 gzserver
-rwxr-xr-x  1 root root   404 Aug 15 20:32 gzclient
-rwxr-xr-x  1 root root   320 Aug 15 20:32 gdbrun
-rwxr-xr-x  1 root root   819 Aug 15 20:32 gazebo
-rwxr-xr-x  1 root root   683 Aug 15 20:32 debug
drwxr-xr-x  2 root root  4096 Feb 12 19:03 .
drwxr-xr-x 89 root root 12288 Feb 12 19:03 ..
'''

Those files sizes look suspiciously small.