ros/rosdistro #10752: resolve potentially conflicting python-vtk rosdep rules

tfoote, 2016-03-17:
We seem to have several python-vtk rules including `python-vtk` `python-vtk5` and pcl has a dependency on `python-vtk6`

I'm opening this ticket to make sure we determine that it's ok, or resolve which ones we should have as rules. If they all don't conflict and can be installed side by side great. But otherwise we probably need to deconflict. 

Recent discussions related:
- https://github.com/ros/rosdistro/pull/10740
- https://github.com/ros-perception/perception_pcl/issues/44#issuecomment-198082216
- https://github.com/ros/rosdistro/pull/10747

vrabaud, 2016-04-10:
In Kinetic, OpenCV3 cannot be installed along PCL right now but:

- OpenCV only needs a runtime dependency on VTK.
- VTK 6 would work so we can change the key in rosdep (and specify a version in REP 003)

Pros for VTK6 by default:
- The latest VTK 5 is from 2012 (cf http://www.vtk.org/doc/release/5.10/html/) and VTK 6 got released in 2013 (https://blog.kitware.com/whats-new-in-vtk-6/)
- VTK7 even got released 2 months ago

Cons:
- Ubuntu's default is still VTK5: http://packages.ubuntu.com/source/xenial/vtk

tfoote, 2016-04-14:
I think that we need to go with libvtk6-dev to be compatible with libpcl. 

Note that the link above connecting `vtk` to `libvtk-dev` is just the source package name. There's a second source package `vtk6` http://packages.ubuntu.com/source/xenial/vtk6

Looking at the `rdepends` (below) it does not look like there are any major things which require `libvtk5-dev`

```
root@12bf7e726429:/# apt-cache rdepends libvtk5-dev
libvtk5-dev
Reverse Depends:
  libvtk5-qt4-dev
  vtk-examples
  vtk-doc
  science-viewing-dev
  science-viewing
  med-imaging-dev
  libvtk6-dev
  libvtk6-dev
  libvtk5.10
  libvtk5-qt4-dev
  libvtk-java
  libvtk5-qt4-dev
root@12bf7e726429:/# apt-cache rdepends libvtk6-dev
libvtk6-dev
Reverse Depends:
  libvtk6-qt-dev
  vtk6-examples
  vtk6-doc
  science-viewing-dev
  libvtk6-qt-dev
  libvtk6-java
  libvtk6-qt-dev
  libpcl-dev
```

jacquelinekay, 2016-04-14:
also @vrabaud can you be more specific about the package conflict you are running into in the current state of things? I was able to install `libpcl-dev` and `ros-kinetic-opencv3` alongside each other, maybe you're referring to a different opencv package that pulls in `libvtk5-dev`--it seems that `ros-kinetic-opencv3` depends only on `libvtk5`. I'm all for going with `libvtk6-dev` if it's necessary but I just wanted to check

vrabaud, 2016-04-14:
I am on Xenial maybe that's why; the conflict is a standard package conflict.

vrabaud, 2016-04-14:
ok, releasing OpenCV3 with the new rosdep and ARM fix. Good night !

tfoote, 2016-05-20:
immediate conflict resolved replaced by #11488 for high level tracking

closed 2016-05-20 by tfoote