ros/rosdistro #10787: SHA1 is deprecated since apt 1.2.7

wjwwood, 2016-03-19:
This shows up on Xenial when doing an `apt-get update` with our repositories (which have a sha1 key):

```
william@ubuntu:~$ sudo apt-get update
Hit:1 http://repositories.ros.org/ubuntu/testing xenial InRelease
Get:2 http://us.archive.ubuntu.com/ubuntu xenial InRelease [116 kB]                                                                   
Hit:3 http://security.ubuntu.com/ubuntu xenial-security InRelease           
Hit:4 http://us.archive.ubuntu.com/ubuntu xenial-updates InRelease
Hit:5 http://us.archive.ubuntu.com/ubuntu xenial-backports InRelease
Get:6 http://us.archive.ubuntu.com/ubuntu xenial/universe amd64 Packages [7,240 kB]
Get:7 http://us.archive.ubuntu.com/ubuntu xenial/universe i386 Packages [7,220 kB]                                                     
Fetched 14.6 MB in 13s (1,099 kB/s)                                                                                                    
Reading package lists... Done
W: gpgv:/var/lib/apt/lists/repositories.ros.org_ubuntu_testing_dists_xenial_InRelease: The repository is insufficiently signed by key 421C365BD9FF1F717815A3895523BAEEB01FA116 (weak digest)
```

SHA1 was deprecated in apt 1.2.7:

From: https://launchpad.net/debian/+source/apt/1.2.7
>   * Do not consider SHA1 usable
  * Test that SHA1-only .diff/Index files are not used
  * test: Use SHA512 digests for GPG, reject SHA1-based signatures

The process for changing this is less clear. I think we just have to:

- Generate a new pgp key
- Upload the public key to the keyserver at ha.pool.sks-keyservers.net (see: http://wiki.ros.org/jade/Installation/Ubuntu#jade.2BAC8-Installation.2BAC8-Sources.Set_up_your_keys)
- Upload the private key to our private build farm config repository
- Change the configurations use the new key
- Tell everyone to add the new key

This needs to be coordinated with the community because as soon as we deploy the new key everyone using only the old key will not be able to do `apt-get update`.

scpeters, 2016-03-31:
Can we tell people to get the new key from the keyserver before deploying it? This could provide uninterrupted service to those who are proactive.

tfoote, 2016-04-01:
Another option would be to deploy parallel instances of our repositories with the new key using a more standard naming layout. 

Like ubuntu/main  ubuntu/testing  instead of ros/ubuntu and ros-shadow-fixed/ubuntu  We could define a period when we'll double index and then after that stop updates. 

Or instead of duplicating we could simply leave the old distros using the old repo, and the new ones could use the new repo, Aka switch kinetic to the new repo url, but leave jade and older on the old one. Once we stop building for indigo and jade that repo will just be frozen. Similarly we've considered moving old content to a different server like Ubuntu does with old-releases vs archive and the old-releases server is not highly mirrored with as much bandwidth.

tfoote, 2016-05-23:
The fix was to change the default certificate and digest algorithm on the server: 

```
cert-digest-algo SHA256
digest-algo SHA256
```

The existing key is now happy having manually deployed this. I've ticketed this to be integrated into the buildfarm_deployment at: https://github.com/ros-infrastructure/buildfarm_deployment/issues/130

Closing this here as it's resolved for the current deployment.

closed 2016-05-23 by tfoote