ros/rosdistro #10822: decision_making on indigo

carlosjoserg, 2016-03-21:
Hi there, not sure if this is the right place to ask.

Does anybody know if there is any reason why the [`decision_making`](http://wiki.ros.org/decision_making) package is not available on the indigo distribution?

I know there is `SMACH`, but I prefer the `decision_ making` approach, and given that it was already on [hydro distro](https://github.com/ros/rosdistro/blob/master/hydro/distribution.yaml#L1318-L1322), I was wondering if it'd be possible to have it on indigo too.

I've been using it for a while via source on indigo... I'm an user, not a developer, but it'd be useful to have it on synaptics.

tfoote, 2016-03-21:
Which distro to release into is up to the maintainer. I suggest opening a ticket for that package https://github.com/cogniteam/decision_making

closed 2016-03-21 by tfoote

carlosjoserg, 2016-03-21:
Thanks, and sorry for the issue!