ros/rosdistro #10914: ffmpeg is not available on debian jessie

tfoote, 2016-03-28:
There's a note here: https://wiki.debian.org/ffmpeg
And it's in jessie backports: https://packages.debian.org/search?searchon=names&keywords=ffmpeg

I think this might be a good time to leverage our bootstrap repo and bring in the debian jessie backport since it's available on all versions before and after. 

This is breaking the opencv3 builds: http://54.183.26.131:8080/view/Kbin_dJ64/job/Kbin_dJ64__opencv3__debian_jessie_amd64__binary/16/console

wjwwood, 2016-03-28:
I guess `libav` doesn't work in this case? It seems like they recommend it over `ffmepg` in the wiki you linked.

I suppose back porting it is ok, whatever you think is appropriate.

tfoote, 2016-03-28:
It seems that several projects switched to libav but are now in the process of switching back to ffmpeg. See: http://superuser.com/a/941670 and http://stackoverflow.com/a/9477756 I'll look into adding an import of the backport.

tfoote, 2016-03-30:
I added a config here: https://github.com/ros-infrastructure/reprepro-updater/blob/master/config/ffmpeg_debian_jessie_backport.yaml 

And successfully imported on the buildfarm: http://build.ros.org:8080/job/import_upstream/26/console

They will propogate to packages.ros.org with the next syncs.

closed 2016-03-30 by tfoote