ros/rosdistro #10995: Assimp dep on OSX

juanvallejo, 2016-04-04:
Installing `assimp` through brew on OSX does not work with the Jade distribution of ROS. Brew installs the `assimp` version with unified headers but the `catkin_install_isolated` script looks for an older version without the unified headers regardless of any flags set by the `CMakeLists` file in `src/geometric_shapes/CMakeLists.txt`. The version of `assimp` (with no unified headers) that worked for me when installing on a mac was this one: [https://sourceforge.net/projects/assimp/files/assimp-2.0/assimp--2.0.863-sdk.zip/download](https://sourceforge.net/projects/assimp/files/assimp-2.0/assimp--2.0.863-sdk.zip/download). Since this version is `2.0.863`, yet the `CMakeLists.txt` file assumes any version greater than `2.0.1150` has unified headers, I had to patch the `CMakeLists.txt` file in `src/geometric_shapes/CMakeLists.txt` so that it did not set the flag to expect assimp's unified headers after building the above version of `assimp` from source.

wjwwood, 2016-04-04:
It sounds like `geometric_shapes` needs to be patched to correctly compare the versions of `assimp`. I would suggest opening an issue there:

https://github.com/ros-planning/geometric_shapes/issues

wjwwood, 2016-04-04:
@juanvallejo there even look to be some other issues related to building on OS X on `geometric_shapes`. Unfortunately it looks to be under-maintained, but I still think posting an issue there is the best course of action. If someone comes through and cleans up the issues/pull requests against `geometric_shapes` then this issue won't have any visibility to them.

I'm going to close this for now. If you think it affects more packages, then I can reopen this issue. Thanks.

closed 2016-04-04 by wjwwood