ros/rosdistro #11020: Missing packages for Kinetic in Debian Jessie

jacquelinekay, 2016-04-07:
There are some missing packages in Debian Jessie that are causing problems on the Kinetic build farm.

- `libopenni-dev` is not in Jessie for arm.
  - https://packages.debian.org/search?keywords=libopenni-dev
  - This is a dependency of `openni_camera` (of course)
- `collada-dom-dev` is not in Jessie.
  - https://packages.debian.org/search?suite=default&section=all&arch=any&searchon=names&keywords=collada-dom
  - This is a dependency of `collada_parser` and `collada_urdf`.
- `python-wxgtk-2.8` is not Jessie, it has been replaced by python-wxgtk-3.0.
  - https://packages.debian.org/search?keywords=python-wxgtk
  - This is a dependency of `joint_state_publisher`.
  - I should look into migrating joint_state_publisher to the new version, since python-wxgtk-3.0 is supported in Ubuntu Xenial and Wily.

Not sure how to proceed about the first two.

wjwwood, 2016-04-07:
It might also be worth just rewriting `joint_state_publisher` in Qt or PyQt, so we can shed the `wx` dependency. I think it might be the only thing that still hasn't been converted to Qt and I think it is a relatively simple UI.

jacquelinekay, 2016-04-07:
Ticketed here https://github.com/ros/robot_model/issues/67

tfoote, 2016-04-07:
For openni we can try to compile the jessie version from the sourcedeb and import that into our bootstrap repo. So as to keep the same version if possible. If that doesn't work we could look at backporting from sretch

For collada since there's no version in jessie to keep the same behavior we can just import the stretch version.

jacquelinekay, 2016-04-07:
@tfoote How would I go about importing the stretch version of collada for jessie? Would I have to set the rosdep key for jessie in a similar way to this: https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L3767-L3770

Compiling openni for arm sounds nontrivial, I can take a look at it but I'll need some guidance on where to start.

@wjwwood I'll look at using PyQt in `joint_state_publisher` and maybe also resolving ros-visualization/rqt_common_plugins#250. I haven't done Python GUI programming in a while so it will be interesting.

dirk-thomas, 2016-04-07:
@jacquelinekay You might want to use `python_qt_binding` instead. Then your UI will work with either PyQt or PySide.

tfoote, 2016-04-07:
> @tfoote How would I go about importing the stretch version of collada for jessie? Would I have to set the rosdep key for jessie in a similar way to this: https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L3767-L3770

That's a rosdep source install which is an old mechanism to actually have rosdep compile a package locally. For this the rosdep key shouldn't change we want to inject the debian package into the boostrap repository. 

For importing a backport see this config that does it for ffmpeg: https://github.com/ros-infrastructure/reprepro-updater/blob/master/config/ffmpeg_debian_jessie_backport.yaml

> Compiling openni for arm sounds nontrivial, I can take a look at it but I'll need some guidance on where to start.

Yeah, it's basically setup the appropriate target arch image, either docker, VM or hardware. Then download the sourcedeb and invoke dpkg-buildpackage to generate the deb. Then we'd manually import the deb into the repo after verifying it works. I can look at that when I'm back in the office on Tuesday. The difference in version between jessie and stretch looks to be only debian increments and if we can diff those and believe the will both work it would be much easier to just import that version.

jacquelinekay, 2016-04-08:
The OpenCV3 build fails for Debian on arm64 as well. @vrabaud has probably been getting a lot of email spam about this :) The error originates because Protobuf does not recognize the architecture of the build machine.

http://build.ros.org/job/Kbin_arm_dJv8__opencv3__debian_jessie_arm64__binary/
http://build.ros.org/job/Kbin_arm_dJv8__opencv3__debian_jessie_arm64__binary/12/console

This is the macro in Protobuf that checks supported architectures. aarch64 is listed:
https://github.com/google/protobuf/blob/master/src/google/protobuf/stubs/platform_macros.h#L53-L55

Protobuf is in Debian Jessie and available for arm64, so theoretically this should work...
https://packages.debian.org/jessie/libprotobuf-dev

I'm not sure what the problem is here; would our virtualization/containerization scheme break the Protobuf platform detection macro for some reason?

mikaelarguedas, 2017-08-04:
closing this given that kinetic has been out for a while and all dependencies are there

closed 2017-08-04 by mikaelarguedas

deb0ch, 2017-12-14:
I'm still having this issue when trying to follow http://wiki.ros.org/ROSberryPi/Installing%20ROS%20Kinetic%20on%20the%20Raspberry%20Pi and installing the `robot` variant:

```
$ rosdep install -y --from-paths src --ignore-src --rosdistro kinetic -r --os=debian:jessie
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
collada_urdf: No definition of [collada-dom] for OS version [jessie]
collada_parser: No definition of [collada-dom] for OS version [jessie]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```

Tried to install collada-dom using apt:

```
sudo apt-get install libcollada-dom2.4-dp-dev
```
 but did nothing. Also tried instructions from https://answers.ros.org/question/194645/indigo-on-raspberrydebian-dependency-problem-catkin-error/: 

```
cd ~/ros_catkin_ws/external_src
sudo apt-get install libboost-filesystem-dev libxml2-dev
wget http://downloads.sourceforge.net/project/collada-dom/Collada%20DOM/Collada%20DOM%202.4/collada-dom-2.4.0.tgz
tar -xzf collada-dom-2.4.0.tgz
cd collada-dom-2.4.0
cmake .
# ***downgrade gcc***
sudo make install
```

install successful, but did nothing to the error.

what should I do ?