ros/rosdistro #11137: we need to backport colladadom for debian jessie

tfoote, 2016-04-18:
To support Kinetic Re: #9945 

It's failing here: http://build.ros.org/view/Kbin_dJ64/job/Kbin_dJ64__collada_parser__debian_jessie_amd64__binary/17/console
```
23:04:31 Traceback (most recent call last):
23:04:31   File "/usr/lib/python3/dist-packages/apt/cache.py", line 201, in __getitem__
23:04:31     return self._weakref[key]
23:04:31   File "/usr/lib/python3.4/weakref.py", line 125, in __getitem__
23:04:31     o = self.data[key]()
23:04:31 KeyError: 'libcollada-dom2.4-dp-dev'
```

It's available in debian as of stretch: https://packages.debian.org/search?keywords=libcollada-dom2.4-dp0

We can probably build the stretch version for Jessie.

jacquelinekay, 2016-04-18:
Yeah, I tried to update this in ros-infrastructure/reprepro-updater#30 but I didn't realize the backport wasn't available in Jessie. Maybe we can work together on this later today.

jacquelinekay, 2016-05-05:
@tfoote Can we close this?

tfoote, 2016-05-06:
Yes, @j-rivero built colladadom for us and I manually imported it.

closed 2016-05-06 by tfoote