ros/rosdistro #11150: ros organization permissions audit

jacquelinekay, 2016-04-18:
As stated in the email to ros-users, please post in this thread with the permissions you would like to retain for the ros Github organization (ownership and read permissions) before May 2nd.

@ros/owners

jonbinney, 2016-04-18:
Really, I should never have been trusted with any permissions at all......

jonbinney, 2016-04-18:
Actually push is useful for me on some repos (filters, for example, which I occasionally release)

130s, 2016-04-19:
>  the permissions you would like to retain for the ros Github organization (ownership and read permissions)

Is this ticket specifically for `ros` organization (https://github.com/ros) only, or is it also for `ros-*` orgs?

dirk-thomas, 2020-09-01:
Closing due to age and inactivity.

closed 2020-09-01 by dirk-thomas