ros/rosdistro #11360: OS X installation: ogre package is missing

tfabbri, 2016-04-30:
Hello, 

during the installation on Os X of the jade distribution of ROS the package libogre-dev (brew package ogre) is missing. Could you please fix this.

tfoote, 2016-04-30:
What version of OSX are you using? What instructions did you follow? What packages did you install? Please provide enough information to reproduce your problem. And if possible the exact commands your running and their outputs.

tfabbri, 2016-05-01:
Here you can find all the details you asked:

Os version: Yosemite
Instructions followed: http://wiki.ros.org/jade/Installation/OSX/Homebrew/Source
Installation type: Desktop install
Stop step: Resolving dependencies, command:

`$ rosdep install --from-paths src --ignore-src --rosdistro jade -y`

Error message: 

`Error: No available formula with the name "ogre" 
executing command [brew install ogre]
Error: No available formula with the name "ogre" 
==> Searching for similarly named formulae...
These similarly named formulae were found:
homebrew/php/php53-uploadprogress        homebrew/php/php56-uploadprogress      
homebrew/php/php54-uploadprogress        progress                               
homebrew/php/php55-uploadprogress      
To install one of them, run (for example):
  brew install homebrew/php/php53-uploadprogress
==> Searching taps...
These formulae were found in taps:
homebrew/nginx/upload-progress-nginx-module
Caskroom/cask/progressive-downloader
To install one of them, run (for example):
  brew install homebrew/nginx/upload-progress-nginx-module
ERROR: the following rosdeps failed to install
  homebrew: command [brew install ogre] failed `

spaghetti-, 2016-08-10:
Have you tapped `osrf/simulation` which contains the formula that you're missing? You may have forgotten or missed this step.

wjwwood, 2017-01-27:
I believe tapping `orsf/simulation` would be the right fix. I'm going to close this for now. Please comment here and I can reopen if that's not right.

closed 2017-01-27 by wjwwood