ros/rosdistro #11488: Resolve Consistent Qt Versions on Kinetic

tfoote, 2016-05-11:
This is coming out of the ongoing discussion in: https://github.com/ros/rosdistro/pull/11240 however that's a closed PR so can't be reopened. 

Other tickets related include: 
https://github.com/ros/rosdistro/pull/10740
https://github.com/ros-perception/perception_pcl/issues/44#issuecomment-198082216
https://github.com/ros/rosdistro/pull/10747
https://github.com/ros/rosdistro/issues/10752
https://github.com/ros/rosdistro/pull/11095
https://github.com/ros/rosdistro/pull/11097
https://github.com/ros/rosdistro/pull/11240

The high level problem is that we're building libraries which are unrunnable because they're linked against both Qt4 and Qt5. We also have run into problems with mutually exclusive rosdep which cause rosdep to fight back and forth installing conflicting dependencies. 

Complicating this is that based on Xenial we want to have Qt5 be the default version since it's overdue to be deployed and that's what our LTS target supports. 

## Jessie: 
Available: 
* [libqt4-dev ](https://packages.debian.org/jessie/libqt4-dev)
* [qtbase5-dev](https://packages.debian.org/jessie/qtbase5-dev)

Common libraries:
* [libvtk5-dev](https://packages.debian.org/jessie/libvtk5-dev)  depends on [libqt4-dev](https://packages.debian.org/jessie/libqt4-dev)
* [libvtk6-dev](https://packages.debian.org/jessie/libvtk6-dev)  depends on [libqt4-dev](https://packages.debian.org/jessie/libqt4-dev)
* [libpcl-dev](https://packages.debian.org/jessie/libpcl-dev) depends on libvtk5 depends on libqt4-dev
* libgazebo7-dev depends on libqt4-dev

ROS libraries:
* opencv3 Qt5
* rviz Qt5
* rqt Qt5

## Wily:

Very similar to Jessie...
* [vtk[5]](http://packages.ubuntu.com/source/wily/vtk) depends on qt4
* [vtk6](http://packages.ubuntu.com/source/wily/vtk6) depends on qt4
...

## Xenial: 
Available: 
* [libqt4-dev ](http://packages.ubuntu.com/xenial/libqt4-dev)
* [qtbase5-dev](http://packages.ubuntu.com/xenial/qtbase5-dev)

Common libraries:
* [libvtk5-dev](http://packages.ubuntu.com/xenial/libvtk5-dev)  depends on [libqt4-dev](http://packages.ubuntu.com/xenial/libqt4-dev)
* [libvtk6-dev](http://packages.ubuntu.com/xenial/libvtk6-dev)  actually [libvtk6-qt-dev](http://packages.ubuntu.com/libvtk6-qt-dev) depends on [qtbase5-dev](http://packages.ubuntu.com/xenial/qtbase5-dev)
* [libpcl-dev](http://packages.ubuntu.com/xenial/libpcl-dev) depends on libvtk6-dev which depends on qtbase5-dev
* [libgazebo7-dev](http://packages.ubuntu.com/xenial/libgazebo7-dev) depends on libqt4-dev

* [libopencv-dev](http://packages.ubuntu.com/xenial/libopencv-dev) none?

ROS libraries:
* opencv3 Qt5
* rviz Qt5
* rqt Qt5

## Conflict

On Xenial everything is quite hunky dory, everything is Qt5 and things basically work. Except libgazebo7-dev which is not linked against many things except gazebo plugins. 

However a package like depth_image_proc currently builds a nodelet which is linked against both qt4 and qt5 which leads to a segfault at load time. Any other package sufficiently high in the stack to have dependencies from both qt4 and qt5 will crash. For example rviz is currently crashing on startup too. 

## Potential solutions

There's many potential solutions I'm not sure what's the best approach. 

### Backport a lot to make qt5 everywhere?
* backport a ton from stretch into jessie?
* and backport a ton from xenial into wily?
* We'd have to pull in all of vtk and potentially everyone using vtk or Qt might be incompatible. 

### switch jessie and wily to Qt4. 
 * We'd have to support rviz + rqt with Qt4 and 5

### Find some middle ground???
* Figure out how to firewall things to either use Qt5 or Qt4 and not have them overlap like gazebo and rviz.

### Other suggestions?
* Does anyone else have suggestions. 

@jacquelinekay FYI tagged for kinetic

vrabaud, 2016-05-11:
Hi, I fixed depth_image_proc and OpenCV3 so please correct the text above: OpenCV3 now uses whatever Qt that is deemed appropriate by VTK (that means Qt4 on Wily/jessie, Qt5 elewhere).

I do not know of any RViz and OpenCV plugin, RViz and Qt plugin. But I do know of at least one rqt and OpenCV plugin: https://github.com/ros-visualization/rqt_common_plugins/blob/master/rqt_image_view/CMakeLists.txt#L5
and doing an ldd on it on Debian, I do not see any Qt4 (that the dependency on OpenCV3 could have brought in, but the dependencies on OpenCV being non-GUI, the linker was smart enough to not include Qt4) but I do see Qt5 obviously as this is what rqt uses.

So all seems good to me as long as we don't have an RViz or rqt plugin that pops up an OpenCV window (which would therefore not use the RViz or rqt infrastructure .... which would be insane).

Anyway, to me, all looks good and we can probably remove the critical status, but we should definitely keep an eye open.

tfoote, 2016-05-11:
Great, thanks. Testing today I can confirm that my Debian Jessie machine is running the nodelets as well. From: https://github.com/ros-perception/image_pipeline/issues/196

However rviz crashes on startup. And rqt crashes on load of the image_view plugin.

wjwwood, 2016-05-12:
@tfoote do you have anything more substantial than rviz crashes? I don't see why it would be linked against both qt 4 and 5, as it doesn't use vtk or pcl. Also only the visualization part of pcl should be linked against vtk right?

tfoote, 2016-05-12:
Here's what happens when I run rviz on Jessie. 

```
tfoote@plover:~/pcl/stretch$ . /opt/ros/kinetic/setup.bash 
tfoote@plover:~/pcl/stretch$ gdb rviz
GNU gdb (Debian 7.7.1+dfsg-5) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from rviz...(no debugging symbols found)...done.
(gdb) r
Starting program: /opt/ros/kinetic/bin/rviz 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007fffd9557dbc in ?? () from /usr/lib/x86_64-linux-gnu/libQtTest.so.4
(gdb) bt full
#0  0x00007fffd9557dbc in ?? () from /usr/lib/x86_64-linux-gnu/libQtTest.so.4
No symbol table info available.
#1  0x00007ffff7dea9ba in call_init (l=<optimized out>, argc=argc@entry=1, argv=argv@entry=0x7fffffffe1c8, env=env@entry=0x7fffffffe1d8) at dl-init.c:78
        j = <optimized out>
        jm = <optimized out>
        addrs = <optimized out>
        init_array = <optimized out>
#2  0x00007ffff7deaaa3 in call_init (env=0x7fffffffe1d8, argv=0x7fffffffe1c8, argc=1, l=<optimized out>) at dl-init.c:36
No locals.
#3  _dl_init (main_map=0x7ffff7ffe1a8, argc=1, argv=0x7fffffffe1c8, env=0x7fffffffe1d8) at dl-init.c:126
        preinit_array = <optimized out>
        preinit_array_size = <optimized out>
        i = 119
#4  0x00007ffff7ddd1ca in _dl_start_user () from /lib64/ld-linux-x86-64.so.2
        _dl_rtld_libname2 = {name = 0x0, next = 0x0, dont_free = 0}
        start_time = 202066427956217
        _dl_rtld_libname = {name = 0x400200 "/lib64/ld-linux-x86-64.so.2", next = 0x7ffff7ffe010 <newname>, dont_free = 0}
        library_path = 0x7fffffffebf0 "/opt/ros/kinetic/lib"
        audit_list = 0x0
        relocate_time = 2476100664
        any_debug = 0
        preloadlist = 0x0
        version_info = 0
        load_time = 1026543333
        tls_init_tp_called = true
        _rtld_local_ro = {_dl_debug_mask = 0, _dl_osversion = 200711, _dl_platform = 0x7fffffffe499 "x86_64", _dl_platformlen = 6, _dl_pagesize = 4096, _dl_inhibit_cache = 0, _dl_initial_searchlist = {
            r_list = 0x7ffff7f6f7b0, r_nlist = 245}, _dl_clktck = 100, _dl_verbose = 0, _dl_debug_fd = 2, _dl_lazy = 1, _dl_bind_not = 0, _dl_dynamic_weak = 0, _dl_fpu_control = 895, 
          _dl_correct_cache_id = 771, _dl_hwcap = 3219913727, _dl_hwcap_mask = 0, _dl_auxv = 0x7fffffffe358, _dl_inhibit_rpath = 0x0, _dl_origin_path = 0x0, _dl_use_load_bias = 18446744073709551615, 
          _dl_profile = 0x0, _dl_profile_output = 0x7ffff7df5140 "/var/tmp", _dl_trace_prelink = 0x0, _dl_trace_prelink_map = 0x0, _dl_init_all_dirs = 0x7ffff7ff74c8, _dl_hp_timing_overhead = 18, 
          _dl_sysinfo_dso = 0x7ffff7ff8000, _dl_sysinfo_map = 0x7ffff7ffe740, _dl_hwcap2 = 0, _dl_debug_printf = 0x7ffff7deb8d0 <_dl_debug_printf>, _dl_catch_error = 0x7ffff7dea800 <_dl_catch_error>, 
          _dl_signal_error = 0x7ffff7dea5d0 <_dl_signal_error>, _dl_mcount = 0x7ffff7decb70 <_dl_mcount>, _dl_lookup_symbol_x = 0x7ffff7de55e0 <_dl_lookup_symbol_x>, 
          _dl_check_caller = 0x7ffff7dedef0 <_dl_check_caller>, _dl_open = 0x7ffff7dee350 <_dl_open>, _dl_close = 0x7ffff7df0230 <_dl_close>, _dl_tls_get_addr_soft = 0x7ffff7dedaf0 <_dl_tls_get_addr_soft>, 
          _dl_discover_osversion = 0x7ffff7df1550 <_dl_discover_osversion>, _dl_audit = 0x0, _dl_naudit = 0}
        _dl_skip_args = 0
        __pointer_chk_guard_local = 17654171837701726406
        _rtld_global_ro = {_dl_debug_mask = 0, _dl_osversion = 200711, _dl_platform = 0x7fffffffe499 "x86_64", _dl_platformlen = 6, _dl_pagesize = 4096, _dl_inhibit_cache = 0, _dl_initial_searchlist = {
            r_list = 0x7ffff7f6f7b0, r_nlist = 245}, _dl_clktck = 100, _dl_verbose = 0, _dl_debug_fd = 2, _dl_lazy = 1, _dl_bind_not = 0, _dl_dynamic_weak = 0, _dl_fpu_control = 895, 
          _dl_correct_cache_id = 771, _dl_hwcap = 3219913727, _dl_hwcap_mask = 0, _dl_auxv = 0x7fffffffe358, _dl_inhibit_rpath = 0x0, _dl_origin_path = 0x0, _dl_use_load_bias = 18446744073709551615, 
          _dl_profile = 0x0, _dl_profile_output = 0x7ffff7df5140 "/var/tmp", _dl_trace_prelink = 0x0, _dl_trace_prelink_map = 0x0, _dl_init_all_dirs = 0x7ffff7ff74c8, _dl_hp_timing_overhead = 18, 
          _dl_sysinfo_dso = 0x7ffff7ff8000, _dl_sysinfo_map = 0x7ffff7ffe740, _dl_hwcap2 = 0, _dl_debug_printf = 0x7ffff7deb8d0 <_dl_debug_printf>, _dl_catch_error = 0x7ffff7dea800 <_dl_catch_error>, 
          _dl_signal_error = 0x7ffff7dea5d0 <_dl_signal_error>, _dl_mcount = 0x7ffff7decb70 <_dl_mcount>, _dl_lookup_symbol_x = 0x7ffff7de55e0 <_dl_lookup_symbol_x>, 
          _dl_check_caller = 0x7ffff7dedef0 <_dl_check_caller>, _dl_open = 0x7ffff7dee350 <_dl_open>, _dl_close = 0x7ffff7df0230 <_dl_close>, _dl_tls_get_addr_soft = 0x7ffff7dedaf0 <_dl_tls_get_addr_soft>, 
[... 45 lines cut ...]
[... cut ...]
          _dl_initial_dtv = 0x7ffff7f5d190, _dl_tls_generation = 1, _dl_init_static_tls = 0x7ffff6d37ae0 <__pthread_init_static_tls>, _dl_wait_lookup_done = 0x7ffff6d37c00 <__wait_lookup_done>, 
          _dl_scope_free_list = 0x0}
        _dl_argc = 1
        _dl_argv_internal = 0x7fffffffe1c8
#5  0x0000000000000001 in ?? ()
No symbol table info available.
#6  0x00007fffffffe4ac in ?? ()
No symbol table info available.
---Type <return> to continue, or q <return> to quit---
#7  0x0000000000000000 in ?? ()
No symbol table info available.
(gdb) 
```

wjwwood, 2016-05-13:
Looks like ld is failing in `/usr/lib/x86_64-linux-gnu/libQtTest.so.4`, can you post the `ldd` of `rviz` and `librviz_default_plugin`? I don't have a Jessie machine handy.

tfoote, 2016-05-13:
@wjwwood here's the ldd output for both: https://gist.github.com/tfoote/2411ff3db32d401298bb473c71f52460

wjwwood, 2016-05-13:
I setup a Jessie VM, and built rviz from source. Here is the link line for `rviz` itself:

https://gist.github.com/wjwwood/cca1e64d4804334236d7879de31ad7d6

Qt 4 isn't in there, so I guess it is coming in from a dependency. I'll look at each of them.

wjwwood, 2016-05-13:
`/opt/ros/kinetic/lib/libopencv_highgui3.so.3.1.0` links against Qt4 and `rviz

wjwwood, 2016-05-13:
`/opt/ros/kinetic/lib/libopencv_highgui3.so.3.1.0` links against Qt4 and `rviz` links against it. I can look into why rviz is linking against these, but it uses `/opt/ros/kinetic/lib/libimage_geometry.so` also, and that brings in Qt4 too.

I think this might just be an overlinking issue, because I don't think image geometry needs the part of OpenCV that uses Qt and neither does `rviz`.

jacquelinekay, 2016-05-16:
Despite our discussion offline, I still see unaddressed problems on Jessie. pcl or gazebo will conflict with rviz or rqt or OpenCV3. Doesn't that mean that desktop-full will be uninstallable or at least unusable on Jessie?

wjwwood, 2016-05-16:
> Despite our discussion offline, I still see unaddressed problems on Jessie. pcl or gazebo will conflict with rviz or rqt or OpenCV3. Doesn't that mean that desktop-full will be uninstallable or at least unusable on Jessie?

Qt 4 and Qt 5 are side-by-side installable and you can run/build programs which use each (separately), you just cannot have a single application which links against both Qt4/5.

Temigo, 2016-06-29:
@vrabaud : Are your fixes on OpenCV3 available in the current version ? 

I may not have the latest version of opencv3 then, but just in case : the cvv module of opencv_contrib will be built and linked against Qt5 even with the cmake flag WITH_QT=4 (see the CMake file of the module).

tilk, 2016-08-09:
The find_object_2d package also crashes in Qt on Debian Jessie because of the issue:

```
#0  0x00007fffdeca1dbc in ?? () from /usr/lib/x86_64-linux-gnu/libQtTest.so.4
#1  0x00007ffff7dea9ba in call_init (l=<optimized out>, argc=argc@entry=2, 
    argv=argv@entry=0x7fffffffdfa8, env=env@entry=0x7fffffffdfc0)
    at dl-init.c:78
#2  0x00007ffff7deaaa3 in call_init (env=0x7fffffffdfc0, argv=0x7fffffffdfa8, 
    argc=2, l=<optimized out>) at dl-init.c:36
#3  _dl_init (main_map=0x7ffff7ffe1a8, argc=2, argv=0x7fffffffdfa8, 
    env=0x7fffffffdfc0) at dl-init.c:126
#4  0x00007ffff7ddd1ca in _dl_start_user () from /lib64/ld-linux-x86-64.so.2
#5  0x0000000000000002 in ?? ()
#6  0x00007fffffffe2da in ?? ()
#7  0x00007fffffffe30d in ?? ()
#8  0x0000000000000000 in ?? ()
```