ros/rosdistro #11567: debian does not have an equivalent of linux-headers-generic

tfoote, 2016-05-20:
As defined in this rule: https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L2846

Searching for it: 
https://packages.debian.org/search?suite=stable&section=all&arch=any&searchon=names&keywords=linux-headers-generic

This is breaking: http://build.ros.org/view/Kbin_dJ64/job/Kbin_dJ64__schunk_canopen_driver__debian_jessie_amd64__binary/4/console

fmauch, 2016-05-27:
As far as I can see there's currently no option to add architecture dependent rules in the rosdep, right? Because debian seems to only offer version-independent packages for each architecture.
The required packages are then...:
  - linux-headers-amd64
  - linux-headers-486
  - linux-headers-arm64
Also the scripting interface is deprecated and disabled, right? Otherwise the required package could be installed using 
apt-get install linux-headers-$(dpkg --print-architecture)

Does rosdep provide such functionality or would it be necessary to extend rosdep by an architecture flag?

dirk-thomas, 2020-09-01:
Closing due to age and inactivity.

closed 2020-09-01 by dirk-thomas