ros/rosdistro #11583: libpcl-dev missing on Ubuntu armhf

tfoote, 2016-05-20:
It seems that libpcl-dev is packaged on all our target Ubuntu architectures except armhf which explains why this build is failing consistently: http://build.ros.org/view/Kbin_arm_uXhf/job/Kbin_arm_uXhf__pcl_conversions__ubuntu_xenial_armhf__binary/

Upstream package info: http://packages.ubuntu.com/xenial/libpcl-dev
It's available for amd64, arm64, i386. 

It's also available in debian for armhf: https://packages.debian.org/jessie/libpcl-dev https://packages.debian.org/stretch/libpcl-dev

@jspricke @thomas-moulard (I don't know Leo's github username) Do you know why it's not imported for Ubuntu armhf? I'm not sure where to look for past discussions of the state of the import. 

@j-rivero @jacquelinekay FYI

jspricke, 2016-05-20:
It needs a rebuild after vtk is fixed in Ubuntu:
```
make[3]: *** No rule to make target '/usr/lib/arm-linux-gnueabihf/libproj.so', needed by 'lib/arm-linux-gnueabihf/libpcl_io.so.1.7.2'.  Stop.
```
from: https://launchpadlibrarian.net/253786845/buildlog_ubuntu-xenial-armhf.pcl_1.7.2-14build1_BUILDING.txt.gz

zcox, 2016-06-18:
Could this explain why move_base is not available for armhf?

http://answers.ros.org/question/235521/move_base-for-kinetic-armhf/

I tried compiling the navigation stack from source on my raspberry pi 3, but quickly ran into libpcl-dev missing in Ubuntu 16.04 armhf.

tfoote, 2016-06-18:
@zcox yes, if you look at the dependencies of move_base you'll find everything depending on pcl is disabled for arm such as: http://build.ros.org/view/Kbin_arm_uXhf/job/Kbin_arm_uXhf__pcl_conversions__ubuntu_xenial_armhf__binary/

blacllisted here: https://github.com/ros-infrastructure/ros_buildfarm_config/blob/production/kinetic/release-arm-build.yaml

tfoote, 2016-06-23:
@jspricke we've [blacklisted pcl_conversions](https://github.com/ros-infrastructure/ros_buildfarm_config/pull/43) on arm which effectively disables all ros dependencies on pcl trying to build. Is there anywhere we can track the fixes to vtk being released to fix this?

tfoote, 2016-06-23:
I found: 

https://bugs.launchpad.net/ubuntu/+source/pcl/+bug/1573174
Which led me to 
https://bugs.launchpad.net/ubuntu/+source/vtk6/+bug/1573234 Which says it's fixed for yakkety, but needs an SRU for xenial @j-rivero could you see if you could accelerate that?

jspricke, 2016-06-23:
@tfoote you could simply upload PCL to your [PPA](https://launchpad.net/~tully.foote/+archive/ubuntu/backports) and copy the debs. Make sure to follow the version scheme again.

tfoote, 2016-08-10:
I pulled in 1.7.2-14 from: https://anonscm.debian.org/gitweb/?p=debian-science/packages/pcl.git and created an ubuntu xenial changelog on this branch https://github.com/tfoote/pcl-release/commits/rebuild-1.7.2-14

Unfortunately all builds passed except the armhf one with this error:

```
CMakeFiles/pcl_offline_integration.dir/src/opengl_viewer.cpp.o -c /«PKGBUILDDIR»/apps/in_hand_scanner/src/opengl_viewer.cpp
In file included from /«PKGBUILDDIR»/apps/in_hand_scanner/src/opengl_viewer.cpp:52:0:
/usr/include/GL/gl.h:134:17: error: conflicting declaration ‘typedef double GLdouble’
 typedef double  GLdouble; /* double precision float */
                 ^
In file included from /usr/include/qt4/QtOpenGL/QGLWidget:1:0,
                 from /«PKGBUILDDIR»/apps/in_hand_scanner/include/pcl/apps/in_hand_scanner/opengl_viewer.h:46,
                 from /«PKGBUILDDIR»/apps/in_hand_scanner/src/opengl_viewer.cpp:41:
/usr/include/qt4/QtOpenGL/qgl.h:85:17: note: previous declaration as ‘typedef GLfloat GLdouble’
 typedef GLfloat GLdouble;
                 ^
In file included from /usr/include/GL/gl.h:2055:0,
                 from /«PKGBUILDDIR»/apps/in_hand_scanner/src/opengl_viewer.cpp:52:
/usr/include/GL/glext.h:468:19: error: conflicting declaration ‘typedef std::ptrdiff_t GLsizeiptr’
 typedef ptrdiff_t GLsizeiptr;
                   ^
In file included from /usr/include/qt4/QtOpenGL/qgl.h:79:0,
                 from /usr/include/qt4/QtOpenGL/QGLWidget:1,
                 from /«PKGBUILDDIR»/apps/in_hand_scanner/include/pcl/apps/in_hand_scanner/opengl_viewer.h:46,
                 from /«PKGBUILDDIR»/apps/in_hand_scanner/src/opengl_viewer.cpp:41:
/usr/include/GLES2/gl2.h:69:25: note: previous declaration as ‘typedef khronos_ssize_t GLsizeiptr’
 typedef khronos_ssize_t GLsizeiptr;
                         ^
In file included from /usr/include/GL/gl.h:2055:0,
                 from /«PKGBUILDDIR»/apps/in_hand_scanner/src/opengl_viewer.cpp:52:
/usr/include/GL/glext.h:469:19: error: conflicting declaration ‘typedef std::ptrdiff_t GLintptr’
 typedef ptrdiff_t GLintptr;
                   ^
In file included from /usr/include/qt4/QtOpenGL/qgl.h:79:0,
                 from /usr/include/qt4/QtOpenGL/QGLWidget:1,
                 from /«PKGBUILDDIR»/apps/in_hand_scanner/include/pcl/apps/in_hand_scanner/opengl_viewer.h:46,
                 from /«PKGBUILDDIR»/apps/in_hand_scanner/src/opengl_viewer.cpp:41:
/usr/include/GLES2/gl2.h:70:26: note: previous declaration as ‘typedef khronos_intptr_t GLintptr’
 typedef khronos_intptr_t GLintptr;
                          ^
/«PKGBUILDDIR»/apps/in_hand_scanner/src/opengl_viewer.cpp: In member function ‘void pcl::ihs::OpenGLViewer::drawMeshes()’:
/«PKGBUILDDIR»/apps/in_hand_scanner/src/opengl_viewer.cpp:951:61: error: cannot convert ‘const Scalar* {aka const double*}’ to ‘const GLdouble* {aka const float*}’ for argument ‘1’ to ‘void glMultMatrixd(const GLdouble*)’
         glMultMatrixd (mesh.transformation.matrix ().data ());
                                                             ^
/«PKGBUILDDIR»/apps/in_hand_scanner/src/opengl_viewer.cpp: In member function ‘void pcl::ihs::OpenGLViewer::drawBox()’:
/«PKGBUILDDIR»/apps/in_hand_scanner/src/opengl_viewer.cpp:993:59: error: cannot convert ‘Eigen::PlainObjectBase<Eigen::Matrix<double, 4, 4> >::Scalar* {aka double*}’ to ‘const GLdouble* {aka const float*}’ for argument ‘1’ to ‘void glMultMatrixd(const GLdouble*)’
     glMultMatrixd (coeffs.transformation.matrix ().data ());
                                                           ^
make[3]: *** [apps/in_hand_scanner/CMakeFiles/pcl_offline_integration.dir/src/opengl_viewer.cpp.o] Error 1
apps/in_hand_scanner/CMakeFiles/pcl_offline_integration.dir/build.make:147: recipe for target 'apps/in_hand_scanner/CMakeFiles/pcl_offline_integration.dir/src/opengl_viewer.cpp.o' failed
make[3]: Leaving directory '/«PKGBUILDDIR»/build'
make[2]: *** [apps/in_hand_scanner/CMakeFiles/pcl_offline_integration.dir/all] Error 2
CMakeFiles/Makefile2:5291: recipe for target 'apps/in_hand_scanner/CMakeFiles/pcl_offline_integration.dir/all' failed
make[2]: Leaving directory '/«PKGBUILDDIR»/build'
make[1]: *** [all] Error 2
Makefile:163: recipe for target 'all' failed
make[1]: Leaving directory '/«PKGBUILDDIR»/build'
dh_auto_build: make -j1 returned exit code 2
make: *** [build-arch] Error 2
debian/rules:105: recipe for target 'build-arch' failed
dpkg-buildpackage: error: debian/rules build-arch gave error exit status 2
```
From: https://launchpadlibrarian.net/278183927/buildlog_ubuntu-xenial-armhf.pcl_1.7.2-14build10+osrf1_BUILDING.txt.gz

These errors look like the same ones that @jolting experienced [here](https://github.com/ros-perception/perception_pcl/issues/119#issuecomment-237604839) 

And looking deeper into them it appears to be conflicting typedefs in GL headers:
`/usr/include/GL/gl.h:134:17: error: conflicting declaration ‘typedef double GLdouble’
 typedef double  GLdouble; /* double precision float */`
and
`/usr/include/qt4/QtOpenGL/qgl.h:85:17: note: previous declaration as ‘typedef GLfloat GLdouble’
 typedef GLfloat GLdouble;`

jspricke, 2016-08-10:
Looks like: https://bugs.debian.org/798408

tfoote, 2016-08-10:
Thanks @jspricke that definitely looks like the problem. Unfortunately I haven't wrapped my head around exactly how to best work around this. But we do use GLU by default in our opengl rosdep: https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L3122-L3131 which from that thread is not expected to work on arm*. The other relevant rosdep entry is mesa-common-dev: https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L2982-L2985

jolting, 2016-08-11:
No matter how you skin the cat, this is an abuse of a typedef:
```
typedef GLfloat GLdouble;
```
I'm disappointed that Qt wants to do evil things like that. 

With that said, the best way to fix this is not to directly include gl.h in opengl_viewer.cpp since it should only be indirectly using gl through qgl.h.

jspricke, 2016-08-22:
@jolting I tried removing the include, but got the same error.

jspricke, 2016-09-11:
I've change the PCL build dependencies in Debian, so it builds on armhf and armel. Feel free to import it into the ROS archive.

pabloinigoblasco, 2016-09-21:
Any plan to solve this?
Is there any workaround to get the pcl library on ubuntu 16.04 armhf?

tfoote, 2016-09-21:
@jspricke I'm not sure exactly what you're suggesting importing. I found this commit: https://anonscm.debian.org/cgit/debian-science/packages/pcl.git/commit/?id=4854cb9163f6f8cd09dfc4925364c0774c1dd95b  To get this into the ROS repos I'd backport that patch to the 1.7.2 branch and then build it on launchpad?

jspricke, 2016-09-21:
This one:

https://anonscm.debian.org/cgit/debian-science/packages/pcl.git/commit/?id=f511f2e97c64ce082391e4915e86d5f8f4302dd6

pabloinigoblasco, 2016-09-22:
I am trying to follow your steps because I really need any workaround for this. However  I'm getting the classical g++ internal compiler error when there is not enough memory.

How do you compile this for armhf? Do you have any kind of crosscopmiling environment or do you directly build it on a armhf machine? I am working on a OdroidXU4 with 2Gb of RAM. More swap space does not work and I am compiling with -j 1... any idea?

[ 24%] Building CXX object features/CMakeFiles/pcl_features.dir/src/integral_image_normal.cpp.o
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
features/CMakeFiles/pcl_features.dir/build.make:302: recipe for target 'features/CMakeFiles/pcl_features.dir/src/integral_image_normal.cpp.o' failed
make[2]: *** [features/CMakeFiles/pcl_features.dir/src/integral_image_normal.cpp.o] Error 4
CMakeFiles/Makefile2:1005: recipe for target 'features/CMakeFiles/pcl_features.dir/all' failed
make[1]: *** [features/CMakeFiles/pcl_features.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2

jspricke, 2016-09-22:
The Debian buildd used hasse:
https://db.debian.org/machines.cgi?host=hasse which has 4GbB ram.

jspricke, 2016-09-22:
We have a special option for mips, maybe this works or you:

https://anonscm.debian.org/cgit/debian-science/packages/pcl.git/commit/debian/rules?id=a246c2d910c478b75a486c4ebd827f65650ebac9

pabloinigoblasco, 2016-09-22:
> pcl.git/commit/debian/rules?id=a246c2d910c478b75a486c4ebd827f65650ebac9

Thank you for the information.
I will try that.

tfoote, 2016-09-22:
Porting that package back I got a successful build on armhf here: https://launchpad.net/~tully.foote/+archive/ubuntu/backports/+packages It failed on the other platforms for reasons I do not recognize, but I'm only planning to use the armhf platform anyway so won't look into it. 

I've imported those packages into the ros bootstrap repo and they should be available in shadow-fixed momentarily. If anyone has some time to test that would be appreciated.

pabloinigoblasco, 2016-09-22:
Thanks both. Those solutions are enough by the moment for a work around.

@jspricke : your solution to make work the compiler worked like a charm. I will remember the option "--param ggc-min-expand=20". I could compile and use the pcl library.

@tfoote : I downloaded your packages and also work well. Thanks.

tfoote, 2016-09-22:
@pabloinigoblasco thanks for the verification @jspricke thanks for your help throughout the process!

closed 2016-09-22 by tfoote

ErikHells, 2016-09-27:
As an effect of this fix - will the move_base package compile for armhf now? I can't see the move_base in the shadow repo yet.

dirk-thomas, 2016-09-27:
You can check the status page for ARMhf: http://repositories.ros.org/status_page/ros_kinetic_uxhf.html?q=move_base

If you click on the first square you will get to the job which is still waiting to be build: http://build.ros.org/view/Kbin_uxhf_uXhf/job/Kbin_uxhf_uXhf__move_base__ubuntu_xenial_armhf__binary/