ros/rosdistro #11626: libftdipp-dev not resolving properly for Ubuntu 16.04

koenlek, 2016-05-24:
It tries to install "libftdipp-dev" but I think it should be "libftdipp1-dev" on xenial. Should the rosdep be updated from this:

```
libftdipp-dev:
  fedora: [libftdi-c++-devel]
  ubuntu: [libftdipp-dev]
```

To e.g. this:

```
libftdipp-dev:
  fedora: [libftdi-c++-devel]
  ubuntu: 
    precise: [libftdipp-dev]
    trusty: [libftdipp-dev]
    vivid: [libftdipp-dev]
    wily: [libftdipp-dev]
    xenial: [libftdipp1-dev]
```

wjwwood, 2016-05-24:
Please open a pull request to address the problem. You might want to @ mention whoever added it originally to see what they think.

Thanks!

koenlek, 2016-05-29:
Good one. Having some busy days, but I plan to make the pull request when I get the time.

koenlek, 2016-09-30:
Finally found some time ;)

tfoote, 2017-06-08:
Resolved in #12866

closed 2017-06-08 by tfoote