ros/rosdistro #11687: dev-libs/collada-dom and media-libs/collada-dom in gentoo

benkinooby, 2016-05-31:
Greetings,

I tried to install ROS kinect on gentoo but decided to deviate from the official install guide [1][2] occasionally. Most notably I decided against using ros-overlay since all relevant packages seem to be present in portage.

After running
`rosinstall_generator desktop --rosdistro kinetic ...`

the next command fails as follows (other packages were installed too but are not displayed here)
`$ rosdep install --from-paths src --ignore-src --rosdistro kinetic`
`executing command [sudo -H emerge -a media-libs/collada-dom]`
`Password:` 
`These are the packages that would be merged, in order:`
`Calculating dependencies... done!`
`emerge: there are no ebuilds to satisfy "media-libs/collada-dom".`
`emerge: searching for similar names...`
`emerge: Maybe you meant any of these: dev-libs/collada-dom, media-libs/opencollada, media-libs/ladspa-sdk?`
`ERROR: the following rosdeps failed to install`
`portage: command [sudo -H emerge -a media-libs/collada-dom] failed`

After some investigation I found out that `media-libs/collada-dom` is contained in ros-overlay (seems it was created at a point of time where no such packages existed in portage) so I wonder if `dev-libs/collada-dom` could be added to base.yaml now that portages provides it and allow an overlay-free install alltogether (unless there is more to the overlay's existence apart from historical reasons).

Kind regards

Additional information:
William Woodall added the code in question [3] by merging it to base.yaml from gentoo.yaml [4]

[1] http://wiki.ros.org/kinetic/Installation/Gentoo
[2]
- setting python's default version to 2.7 is not recommended and should be covered by gentoo's python-wrapper
- it uses layman which is practically deprecated by the extended functionality of portage (via repos.conf)
- it recommends to run pip install as root which will mess with portage
[3] https://github.com/ros/rosdistro/blame/master/rosdep/base.yaml
[4] https://github.com/ros/rosdistro/blame/52a9adacd4cdc62de793ac1c0f58fa98315b3cc9/rosdep/gentoo.yaml

tfoote, 2016-05-31:
@allenh1 thoughts

allenh1, 2016-06-01:
You make a lot of good points here. The main purpose of the overlay, as I see it, is to keep the builds being stable with one another. Gentoo portage trees have a tendency to fall behind in version numbers, or jump way too far ahead in version numbers. This overlay ensures that ROS users can use compatible builds for ROS installations. Though, yes, conflicts are rare, they certainly happen.

> setting python's default version to 2.7 is not recommended and should be covered by gentoo's python-wrapper

This is true, but ROS is still not fully working with Python 3 (at least as far as I know). So the guide says 2.7 to ensure compatibility with existing ROS packages.

> it uses layman which is practically deprecated by the extended functionality of portage (via repos.conf)

This should definitely be changed in the install guide. Please feel free to update it accordingly.

> it recommends to run pip install as root which will mess with portage

This is also a good point, and is not even a necessary step in the install (I forgot to remove it, and will do so now).

benkinooby, 2016-06-01:
Thanks @allenh1 for your effort.
I'll get to updating the wiki this week or next.

My initial issue for this report still stands though:
collada-dom is in dev-libs (portage) and media-libs (overlay). Having collada-dom in two places when the overlay is used seems suboptimal to me. Possible fixes:

- move collada-dom to dev-libs/collada-dom in the overlay and update base.yaml (probably best solution)
- move collada-dom to media-libs/collada-dom in portage (not likely)
- update base.yaml to allow dev-libs/collada-dom as well as media-libs/collada-dom (cheap and dirty)
 
If there is a good reason to have collada-dom in different places in portage and overlay please state them here for future reference and close this issue.

benkinooby, 2016-06-02:
I had contact with Alexis Ballier, who is a gentoo dev and worked on ROS for gentoo [1].
- He suggested to take it one step further and go with 'ros-meta/desktop' or 'ros-meta/desktop_full' (part of gentoos package management system portages)
- He agrees that base.yaml should be fixed and "media-libs/collada-dom removed from ros-overlay, unless ros-overlay's version adds some missing features to gentoo's portage tree version."

As a consequence I created a pull request [2] which will allow portage to provide collada-dom in order to move away from the overlay.

[1] https://github.com/aballier https://wiki.gentoo.org/wiki/User:Aballier
[2] https://github.com/ros/rosdistro/pull/11724

benkinooby, 2016-06-06:
# INTERMEDIATE WORKAROUND:
Add my base.yaml to your ROS-sources. That will replace the dependency on media-libs/collada-dom with a dependency on dev-libs/collada-dom

1. `# touch /etc/ros/rosdep/sources.list.d/10-fix-collada-dom.list` [1]
2. `# echo "yaml https://raw.githubusercontent.com/benkinooby/rosdistro/ros_from_portage_instead_of_overlay/rosdep/base.yaml" >> /etc/ros/rosdep/sources.list.d/10-fix-collada-dom.list`
3. `$ ros update`

[1] Please note that the filename is chosen in a way that ensures precedence of my file.

SvenWe, 2016-09-24:
A small improvement to @benkinooby's workaround (https://github.com/ros/rosdistro/issues/11687#issuecomment-223975311): You are able to use "file://" links in the `sources.list.d/*.list` files.
So I used the following:

1. `echo "yaml file:///etc/ros/rosdep_hacks/collada-dom_fix.yaml" > /etc/ros/rosdep/sources.list.d/10-collada-dom_fix.list`
2. Content of "/etc/ros/rosdep_hacks/collada-dom_fix.yaml":
```
collada-dom:
  gentoo:
    portage:
      packages: [dev-libs/collada-dom]
```

The reasoning mentioning this, is the fact that the current base.yaml has also a very similar problem regarding pkg-config: It uses "dev-util/pkgconf", while I have "dev-util/pkgconfig" installed (notice the additional two letters "…ig"). So I had to write another fix for that. The situation was that I would have to explicitly disable pkconf's "pkg-config" use flag to merge this package, which seems to do the same job as "pkgconfig". Imho "virtual/pkgconfig" seems to be the better dependency. @allenh1, you changed the previous "dev-util/pkgconfig" to "dev-util/pkgconf" in https://github.com/ros/rosdistro/commit/d29bd69cd75a33ee325637398712e870afa88b74. Do you mind changing it to the virtual package? (Sorry for hijacking this issue a bit.)

allenh1, 2016-09-24:
@SvenWe That is a sensible change.

Alessandro-Barbieri, 2017-06-07:
@benkinooby @SvenWe nice workaround
@tfoote this issue is obsolete since https://github.com/ros/rosdistro/pull/15101

tfoote, 2017-06-07:
Great. Thanks for noticing. I'll close this out.

closed 2017-06-07 by tfoote