ros/rosdistro #11717: libglew for Ubuntu 16.04 Xenial Xerus

Yelen719, 2016-06-02:
In order to install GLEW library on Ubuntu 16.04 Xenial Xerus with rosdep, the libglew-dev section in https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml should be updated to:

```
libglew-dev:
  arch: [glew]
  debian: [libglew-dev]
  fedora: [glew-devel]
  gentoo:
    portage:
      packages: [media-libs/glew]
  ubuntu:
    oneiric: [libglew1.5-dev]
    precise: [libglew1.6-dev]
    quantal: [libglew-dev]
    raring: [libglew-dev]
    saucy: [libglew-dev]
    trusty: [libglew-dev]
    utopic: [libglew-dev]
    vivid: [libglew-dev]
    xenial: [libglew-dev]
```

reference: https://launchpad.net/ubuntu/xenial/+package/libglew-dev

wjwwood, 2016-06-02:
Sounds reasonable. Please open a pull request if you can.

Yelen719, 2016-06-02:
@wjwwood Ok. I'll do it later today.

tfoote, 2017-01-25:
@Yelen719 I'm going to close this. If you have time for the PR that would be great.

closed 2017-01-25 by tfoote