ros/rosdistro #11763: Fedora 24 and gazebo

MalcolmMielle, 2016-06-06:
Ok first of all tell me if it's the wrong package to post the issue.

I'm trying to install kinetic on Fedora 24 (I know it's a beta release) and I have this when trying to install the dependencies : 
```
rosdep install --from-paths src --ignore-src --rosdistro kinetic -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
gazebo_ros: No definition of [libgazebo7-dev] for OS [fedora]
gazebo_plugins: No definition of [libgazebo7-dev] for OS [fedora]
```

I found [this thread](https://github.com/ros/rosdistro/pull/11288) that seems to indicate that the gazebo dependency was removed because Fedora 23-24 do not have Gazebo7 yet and it seems to be a somewhat similar problem.

tfoote, 2016-06-06:
Assuming gazebo7 is still not available on Fedora 24 this looks like the correct message. If it is added to the Fedora repos we can add a rule. Otherwise I suggest trying to work with Fedora maintainers to get gazebo7 released into F24. 

Otherwise you'll want to manually install it on your system and use the `--skip-keys` option or add a local rosdep rule with an empty list to indicate it's already installed.

tfoote, 2017-01-25:
Gazebo 6 appears to be in Fedora 24 while Gazebo7 is in Fedora 25. https://apps.fedoraproject.org/packages/gazebo

You might be able to make it work with a custom build and skipping the keys, but I don't think there's a rosdep rule we can apply that will fix F24.

closed 2017-01-25 by tfoote