ros/rosdistro #12105: libceres-dev missing cmake module

jacquelinekay, 2016-07-08:
quoting @mikepurvis from the buildsystem sig:

"Looks like another upstream packaging snafu— previous to Xenial, ceres-solver was imported into packages.ros.org from Mike Ferguson's PPA:

https://github.com/ros-infrastructure/reprepro-updater/blob/master/config/ceres.upstream.yaml

But now it's in upstream, yay: http://packages.ubuntu.com/xenial/libceres-dev

Except the upstream package is missing its CMake find module, fail: https://bugs.launchpad.net/ubuntu/+source/ceres-solver/+bug/1596296

The problem is in the yakkety deb too. The sid version is the same too: https://packages.qa.debian.org/c/ceres-solver.html

Do we want to patch this in a PPA and pull in the patched package? Ship the module in cmake_modules? Something else?"

tfoote, 2016-07-08:
Now that it's upstream I'd rather not be repackaging it. It will be just adding the files, but maintaining full compatibility is always a challenge and very hard to verify. 

Two options I see:

* One is to inject a small package and add it to the rosdep rule which installs the Find Module on top. 
* Two is to add the rule to the cmake_modules package.

The cmake_modules is a more generic solution and will work on more platforms. However it looks to me like this is a debian packaging issue and that libceres has the appropriate find rules in the upstream sources thus this should not be a problem on the other platforms. Thus I would advocate for approach one. It also has the advantage of being able to be phased out easily once the upstream is fixed to include the find modules. (As I don't expect that to be backported...)

mikepurvis, 2016-07-08:
Issue is in the install manifest:

```
$ dget http://archive.ubuntu.com/ubuntu/pool/universe/c/ceres-solver/ceres-solver_1.11.0~dfsg0-2.dsc
$ cat ceres-solver-1.11.0~dfsg0/debian/libceres-dev.install
include/ceres usr/include
usr/lib/lib*.so
```

So I could add another package for just the module, and then you can pull only that one deb into `packages.ros.org`.

tfoote, 2016-07-08:
> So I could add another package for just the module, and then you can pull only that one deb into packages.ros.org.

Yeah, that's my thought. And we'll modify the rosdep rules to pull both the upstream package and the modules package on the affected distros.

mikepurvis, 2016-07-08:
PPA: https://launchpad.net/~mikepurvis/+archive/ubuntu/ceres

Diff: https://launchpadlibrarian.net/271806013/ceres-solver_1.11.0~dfsg0-2_1.11.0~dfsg0-2cmakefix0.diff.gz

Testing now, will post when all clear. You'll probably want a yakkety package too, eh?

pjreed, 2017-03-27:
It's worth noting that the Debian packages provided by that PPA still do not work because they don't include the header `ceres/internal/config.h`.  Trying to include ceres.h will result in an compilation error like so:

```
In file included from /usr/include/ceres/types.h:42:0,
                 from /usr/include/ceres/internal/variadic_evaluate.h:38,
                 from /usr/include/ceres/internal/autodiff.h:148,
                 from /usr/include/ceres/autodiff_cost_function.h:132,
                 from /usr/include/ceres/ceres.h:37,
                 from main.cpp:15:
/usr/include/ceres/internal/port.h:37:35: fatal error: ceres/internal/config.h: No such file or directory
```

This bug is also tracked on launchpad here: https://bugs.launchpad.net/ubuntu/+source/ceres-solver/+bug/1595692

ahundt, 2017-08-11:
Sorry for hijacking but, where do you put in requests for this sort of thing for updated packages in ubuntu releases? 

I want to nudge the next Ubuntu LTS to have an up to date version of libraries that are highly used but they might not update unless given a reason. 

examples:
cmake, boost, eigen3, ceres-solver, opencv, opencv-contrib, protobuf, flatbuffers, suite-sparse, (other blas libs), zeromq, bullet, googletest, spdlog, openni2.

mikaelarguedas, 2017-08-11:
For Xenial Yakkety and Zesty, we now provide debs for ceres-solver 1.12.0 on `packages.ros.org`. Looks like the only remaining platform with a faulty `ceres-solver` package is Debian Jessie. The upstream package on Debian Stretch works fine and didnt need to be repackaged.

@ahundt AFAIK The best way to get a package in the next ubuntu version (LTS or not) is to have it in debian sid at the time ubuntu imports all debian packages for a distribution (I don't know what that date is for ubuntu 18.04). I can't tell for each of the packages you are referring to but looking at the status of these packages on debian sid and helping the Debian maintainers to package the version you want seems the best way to go.

tfoote, 2017-08-11:
The release schedule for Ubuntu 18.04 has not been defined yet. But I expect it will follow the [17.04 one](https://wiki.ubuntu.com/ZestyZapus/ReleaseSchedule) relatively closely, just a year later plus or minus a week or two. So the import freeze is in February. But I'd definitely suggest starting asap if there are specific things out of date.

Since we have ceres 1.12 for [all Lunar targets](https://discourse.ros.org/t/ceres-package/872/19?u=tfoote). And we're [close on Kinetic](https://discourse.ros.org/t/ceres-package/872/19?u=tfoote) but Jessie is [blocked by dependency issues](https://discourse.ros.org/t/ceres-package/872/20?u=tfoote) I'm going to close this as resolved for Lunar and stop tracking this for a backport to earlier rosdistros. If someone can resolve the dependency issues for the backport to Jessie, we can then support Kinetic releases too. Since high enough versions appear to be upstream, it shouldn't be an issue going forward.

closed 2017-08-11 by tfoote