ros/rosdistro #12528: Help needed to manage a major package reorg.

dustingooding, 2016-08-29:
I'm the new maintainer for the Robonat2 Gazebo simulation and all the related packages that it needs.  We've got new package names, new package repositories, new package metadata, etc.  Old repositories are deprecated and should not be used.

I don't expect the ROS build farm to generate debs from this new codebase, but I do want the ROS wiki to display the most-current information.

Can I get some guidance on where to start?  I'm very new to this.  @scpeters sent me here.

Here's a listing of the new repos and their locations (vcs export).  https://gitlab.com/nasa-jsc-robotics/r2_workspaces/blob/develop/public_sim.yaml

```
repositories:
    fsm_utils:
        type: git
        url: git@gitlab.com:nasa-jsc-robotics/fsm_utils.git
    hayai:
        type: git
        url: git@gitlab.com:nasa-jsc-robotics/hayai.git
    iss_taskboard_gazebo:
        type: git
        url: git@gitlab.com:nasa-jsc-robotics/iss_taskboard_gazebo.git
    nasa_common_cmake:
        type: git
        url: git@gitlab.com:nasa-jsc-robotics/nasa_common_cmake.git
    nasa_common_logging:
        type: git
        url: git@gitlab.com:nasa-jsc-robotics/nasa_common_logging.git
    r2_description:
        type: git
        url: git@gitlab.com:nasa-jsc-robotics/r2_description.git
    r2_gazebo:
        type: git
        url: git@gitlab.com:nasa-jsc-robotics/r2_gazebo.git
    r2_gazebo_gripper:
        type: git
        url: git@gitlab.com:nasa-jsc-robotics/r2_gazebo_gripper.git
    r2_gazebo_interface:
        type: git
        url: git@gitlab.com:nasa-jsc-robotics/r2_gazebo_interface.git
    r2_moveit_config:
        type: git
        url: git@gitlab.com:nasa-jsc-robotics/r2_moveit_config.git
    r2_msgs:
        type: git
        url: git@gitlab.com:nasa-jsc-robotics/r2_msgs.git
    r2_supervisors_control:
        type: git
        url: git@gitlab.com:nasa-jsc-robotics/r2_supervisors_control.git
    r2_upperbody_moveit_config:
        type: git
        url: git@gitlab.com:nasa-jsc-robotics/r2_upperbody_moveit_config.git
    robodyn_controllers:
        type: git
        url: git@gitlab.com:nasa-jsc-robotics/robodyn_controllers.git
    robodyn_mechanisms:
        type: git
        url: git@gitlab.com:nasa-jsc-robotics/robodyn_mechanisms.git
    robodyn_ros:
        type: git
        url: git@gitlab.com:nasa-jsc-robotics/robodyn_ros.git
    robodyn_utilities:
        type: git
        url: git@gitlab.com:nasa-jsc-robotics/robodyn_utilities.git
    robot_instance:
        type: git
        url: git@gitlab.com:nasa-jsc-robotics/robot_instance.git
```

wjwwood, 2016-09-02:
I think you just need to remove the old repository entries in the distribution files, then add new ones with "doc" entries containing the above information. If you do that in one pull request then it should pass, we can merge it, and in a few days (around 4 actually) the docs should be updated on the wiki.

dirk-thomas, 2016-09-02:
In which ROS distro are the current entries listed? I am just asking because anything older then Indigo is not being updated anymore.

dustingooding, 2016-09-02:
They're currently listed in fuerte, groovy, and hydro.  I've targeted the new sim builds for Indigo+.  If I just add new repo entries with the 'doc' parameter, will that only make wiki pages?  I don't want debians built from these (yet).

dirk-thomas, 2016-09-02:
The build farm also does not generate documentation for Hydro and older anymore. So updating those entries won't do anything. We usually don't even allow changes to these distro files anymore.

You can register the new repos with doc entries (using https urls, not "git@") for Indigo and newer.

dustingooding, 2016-09-02:
ok, so don't change <hydro, and do change >indigo.  👍 

as it stands, i'm just adding all those repos listed above one-at-a-time into the indigo/distribution.yaml file.  i don't have a top-level meta that collects them all.  do i need that?

dirk-thomas, 2016-09-02:
> ok, so don't change indigo.

I am not sure if I get that. I would say don't change Hydro and older.

You can add multiple entries in a single PR. And no, you don't need a metapackage.

dustingooding, 2016-09-02:
Alright, I'll go ahead and close this.  If there's any problems with the wiki docs it generates, I'll ask specifically about them.  Thanks!

closed 2016-09-02 by dustingooding