ros/rosdistro #12716: rosdep update fails on Neon (Ubuntu fork)

breznak, 2016-09-15:
ROS Kinetic

Steps to reproduce:
`rosdep update`

Result

```
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index.yaml
Add distro "groovy"

ERROR: Rosdep experienced an error: Could not detect OS, tried ['windows', 'ubuntu', 'slackware', 'rhel', 'qnx', 'osx', 'opensuse', 'opensuse', 'mint', 'linaro', 'gentoo', 'funtoo', 'freebsd', 'fedora', 'elementary', 'debian', 'cygwin', 'centos', 'arch']
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.5

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 351, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 359, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 545, in command_update
    error_handler=update_error_handler)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 466, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/lib/python2.7/dist-packages/rosdep2/gbpdistro_support.py", line 147, in get_gbprepo_as_rosdep_data
    ctx = create_default_installer_context()
  File "/usr/lib/python2.7/dist-packages/rosdep2/__init__.py", line 85, in create_default_installer_context
    m.register_platforms(context)
  File "/usr/lib/python2.7/dist-packages/rosdep2/platforms/debian.py", line 50, in register_platforms
    register_linaro(context)
  File "/usr/lib/python2.7/dist-packages/rosdep2/platforms/debian.py", line 64, in register_linaro
    (os_name, os_version) = context.get_os_name_and_version()
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 112, in get_os_name_and_version
    os_name = self.os_detect.get_name()
  File "/usr/lib/python2.7/dist-packages/rospkg/os_detect.py", line 621, in get_name
    self.detect_os()
  File "/usr/lib/python2.7/dist-packages/rospkg/os_detect.py", line 591, in detect_os
    raise OsNotDetected("Could not detect OS, tried %s"%attempted)
OsNotDetected: Could not detect OS, tried ['windows', 'ubuntu', 'slackware', 'rhel', 'qnx', 'osx', 'opensuse', 'opensuse', 'mint', 'linaro', 'gentoo', 'funtoo', 'freebsd', 'fedora', 'elementary', 'debian', 'cygwin', 'centos', 'arch']
```

System:
My OS is KDE Neon - which is a distribution building atop of Ubuntu, so it's 100% Ubuntu compatible. 

Contents of `/etc/os-release`
```
NAME="KDE Neon"
VERSION="5.7"
ID=neon
ID_LIKE="ubuntu debian"
PRETTY_NAME="KDE neon User Edition 5.7"
VERSION_ID="16.04"
HOME_URL="http://neon.kde.org/"
SUPPORT_URL="http://neon.kde.org/"
BUG_REPORT_URL="http://bugs.kde.org/"
VERSION_CODENAME=xenial
UBUNTU_CODENAME=xenial
```

Proposal:
the test for name in python should parse `ID_LIKE` field.

wjwwood, 2016-09-15:
Thanks for the report.

This should probably be filed on the rosdep issues tracker or maybe the rospkg issue tracker (since rospkg is responsible for detecting OS). I believe there is some discussion on either one of those issue trackers about how to handle rebranded OSes like this one.

https://github.com/ros-infrastructure/rosdep/issues
https://github.com/ros-infrastructure/rospkg/issues

tfoote, 2016-09-15:
To support Neon: 

rospkg will need to be extended to successfully detect it. (You can likely use the [LSB detect module like Elementary](https://github.com/ros-infrastructure/rospkg/blob/master/src/rospkg/os_detect.py#L658))
and rosdep will need to be extended to show that Neon is an alias to Ubuntu. [Here's how Elementary does that](https://github.com/ros-infrastructure/rospkg/issues/25)

I'm going to close this ticket as there's nothing we need to do in the rosdistro. Please open separate issues on those trackers referencing each other as well as here. Note that since Neon is not a commonly used platform we need to rely on community contributions to add support and testing so pull requests that you've tested will greatly accelerate the process.

closed 2016-09-15 by tfoote