ros/rosdistro #12803: ecl_* doc and source branches are invalid

tfoote, 2016-09-22:
It looks like the repo has been updated in it's layout and `indigo-devel` and `indigo-stable` branches are no longer valid. 

For example: https://github.com/ros/rosdistro/blob/master/indigo/distribution.yaml#L2188 and https://github.com/ros/rosdistro/blob/master/indigo/distribution.yaml#L2221

It also looks like a problem for jade and kinetic entries. 

@stonier could you take a look?

stonier, 2016-09-25:
Should be fixed by:

* [indigo](https://github.com/ros/rosdistro/pull/12819/files)
* [jade](https://github.com/ros/rosdistro/pull/12821)
* [kinetic](https://github.com/ros/rosdistro/pull/12820)

Thanks for the heads up.

stonier, 2017-02-21:
@tfoote this can be safely closed now.

closed 2017-02-21 by tfoote