ros/rosdistro #12824: prerelease.sh fails to clone ros_buildfarm

pirobot, 2016-09-25:
Apologies if this is not the right place to post this issue but I couldn't find an issues URL for creating releases.

My setup is Ubuntu 14.04 and ROS Indigo debian install.  I have docker installed according to the instructions on the docker website and I have Python3 and EmPy installed.

I am running through the [Prerelease Test instructions](http://wiki.ros.org/bloom/Tutorials/PrereleaseTest) (see also the [prerelease setup guide](http://wiki.ros.org/regression_tests#How_do_I_setup_my_system_to_run_a_prerelease.3F)) and made it almost all the way through except that when running the prerelease.sh script, a fatal error occurs in the prerelease_build_underlay.sh sub-script at the line:

git -C ros_buildfarm log -n 1

because the ros_buildfarm directory does not exist.  Looking at the previous line in the script:

python3 -u wrapper_scripts/git.py clone --depth 1 -b 1.1.0 https://github.com/ros-infrastructure/ros_buildfarm.git ros_buildfarm

I ran this command on its own and it generates no output, no error, and does not create the ros_buildfarm directory.  I then replaced this command with:

git clone --depth 1 -b 1.1.0 https://github.com/ros-infrastructure/ros_buildfarm.git ros_buildfarm

and everything worked fine.  This same command appears in the prerelease_build_overlay.sh and it fails in the same way and the same fix works.

dirk-thomas, 2016-09-25:
The repo https://github.com/ros-infrastructure/ros_buildfarm would be a better fit for the ticket but we can keep the issue here. I tried to reproduce your case but it works fine for me. When you say:

> I ran this command on its own and it generates no output, no error, and does not create the ros_buildfarm directory.

that sounds surprising to me. I would expect at least some output. Can you please check that you actually have the file `wrapper_scripts/git.py`. It should be generated by the `prerelease_build_underlay.sh` script - one step above the `git clone`. For completeness can you please post the exact command you are invoking as well as the full output of the `./prerelease.sh`. The output might provide some information about what is going wrong.

pirobot, 2016-09-26:
The file wrapper_scripts/git.py *is* created and I list its contents below.  I also list the output from the ./prerelease.sh command.  If, after running ./prerelease.sh, I run the command:

`python3 -u wrapper_scripts/git.py clone --depth 1 -b 1.1.0 https://github.com/ros-infrastructure/ros_buildfarm.git ros_buildfarm`

there is no output at all and the ros_buildfarm directory is not created.

Here is the wapper_scripts/git.py file that gets created:

`#! /usr/bin/python\n\nimport subprocess\nimport sys\nfrom time import sleep\n\n\ndef main(argv=sys.argv[1:]):\n    max_tries = 10\n    known_error_strings = [\n        'Connection timed out',\n    ]\n\n    command = argv[0]\n    if command == 'clone':\n        rc, _, _ = call_git_repeatedly(\n            argv, known_error_strings, max_tries)\n        return rc\n    else:\n        assert "Command '%s' not implemented" % command\n\n\ndef call_git_repeatedly(argv, known_error_strings, max_tries):\n    command = argv[0]\n    for i in range(1, max_tries + 1):\n        if i > 1:\n            sleep_time = 5 + 2 * i\n            print("Reinvoke 'git %s' (%d/%d) after sleeping %s seconds" %\n                  (command, i, max_tries, sleep_time))\n            sleep(sleep_time)\n        rc, known_error_conditions = call_git(argv, known_error_strings)\n        if rc == 0 or not known_error_conditions:\n            break\n        print('')\n        print('Invocation failed due to the following known error conditions: '\n              ', '.join(known_error_conditions))\n        print('')\n        # retry in case of failure with known error condition\n    return rc, known_error_conditions, i\n\n\ndef call_git(argv, known_error_strings):\n    known_error_conditions = []\n\n    cmd = ['git'] + argv\n    print("Invoking '%s'" % ' '.join(cmd))\n    proc = subprocess.Popen(\n        cmd, stdout=subprocess.PIPE, stderr=subprocess.STDOUT)\n    while True:\n        line = proc.stdout.readline()\n        if not line:\n            break\n        line = line.decode()\n        sys.stdout.write(line)\n        for known_error_string in known_error_strings:\n            if known_error_string in line:\n                if known_error_string not in known_error_conditions:\n                    known_error_conditions.append(known_error_string)\n    proc.wait()\n    rc = proc.returncode\n    return rc, known_error_conditions\n\n\nif __name__ == '__main__':\n    sys.exit(main())
`

And here is the entire output from running ./prerelease.sh

```
Prerelease script

Use workspace: /tmp/prerelease_job

Clone source repositories for underlay workspace

+ git clone -b indigo-devel https://github.com/pirobot/pi_trees.git catkin_workspace/src/pi_trees
Cloning into 'catkin_workspace/src/pi_trees'...
remote: Counting objects: 410, done.
remote: Total 410 (delta 0), reused 0 (delta 0), pack-reused 410
Receiving objects: 100% (410/410), 65.37 KiB | 0 bytes/s, done.
Resolving deltas: 100% (193/193), done.
Checking connectivity... done.
+ git -C catkin_workspace/src/pi_trees --no-pager log -n 1
commit 005c2bea905c1744ff36a2a24ba946786b3947f2
Author: Patrick Goebel <patrick@pirobot.org>
Date:   Sun Sep 25 08:45:42 2016 -0700

    Updated CHANGELOG files before release

Clone release repositories for overlay workspace

Underlay workspace contains 3 packages:
- pi_trees
- pi_trees_lib
- pi_trees_ros
Overlay packages based on dependency level: 0
Overlay packages based on whitelisted package names: 0
Overlay workspace will contain 0 packages:

Build underlay workspace

Build step 1
+ echo '# BEGIN SECTION: docker version'
# BEGIN SECTION: docker version
+ docker version
Client:
 Version:      1.12.1
 API version:  1.24
 Go version:   go1.6.3
 Git commit:   23cf638
 Built:        Thu Aug 18 05:22:43 2016
 OS/Arch:      linux/amd64

Server:
 Version:      1.12.1
 API version:  1.24
 Go version:   go1.6.3
 Git commit:   23cf638
 Built:        Thu Aug 18 05:22:43 2016
 OS/Arch:      linux/amd64
+ echo '# END SECTION'
# END SECTION
+ echo '# BEGIN SECTION: docker info'
# BEGIN SECTION: docker info
+ docker info
[... 40 lines cut ...]
# BEGIN SECTION: Check docker status
+ echo 'Testing trivial docker invocation...'
Testing trivial docker invocation...
+ docker run --rm ubuntu:trusty true
+ echo ''\''docker run'\'' returned 0'
'docker run' returned 0

Build step 3
+ echo '# BEGIN SECTION: Embed wrapper scripts'
# BEGIN SECTION: Embed wrapper scripts
+ rm -fr wrapper_scripts
+ mkdir wrapper_scripts
+ echo '#! /usr/bin/python\n\nimport subprocess\nimport sys\nfrom time import sleep\n\n\ndef main(argv=sys.argv[1:]):\n    max_tries = 10\n    known_error_strings = [\n        '\''Connection timed out'\'',\n    ]\n\n    command = argv[0]\n    if command == '\''clone'\'':\n        rc, _, _ = call_git_repeatedly(\n            argv, known_error_strings, max_tries)\n        return rc\n    else:\n        assert "Command '\''%s'\'' not implemented" % command\n\n\ndef call_git_repeatedly(argv, known_error_strings, max_tries):\n    command = argv[0]\n    for i in range(1, max_tries + 1):\n        if i > 1:\n            sleep_time = 5 + 2 * i\n            print("Reinvoke '\''git %s'\'' (%d/%d) after sleeping %s seconds" %\n                  (command, i, max_tries, sleep_time))\n            sleep(sleep_time)\n        rc, known_error_conditions = call_git(argv, known_error_strings)\n        if rc == 0 or not known_error_conditions:\n            break\n        print('\'''\'')\n        print('\''Invocation failed due to the follo [... cut ...] turncode\n    return rc, known_error_conditions\n\n\nif __name__ == '\''__main__'\'':\n    sys.exit(main())'
+ echo '# END SECTION'
# END SECTION

Build step 4
+ echo '# BEGIN SECTION: Clone ros_buildfarm'
# BEGIN SECTION: Clone ros_buildfarm
+ rm -fr ros_buildfarm
+ python3 -u wrapper_scripts/git.py clone --depth 1 -b 1.1.0 https://github.com/ros-infrastructure/ros_buildfarm.git ros_buildfarm
+ git -C ros_buildfarm log -n 1
fatal: Cannot change to 'ros_buildfarm': No such file or directory

```

dirk-thomas, 2016-09-26:
It looks like your long `echo` line in build step 3 lacks `> wrapper_scripts/git.py` at the end. Did you copy the content exactly? I can't see a reason for that line to be different.

Your `wapper_scripts/git.py` file contain the literal `\n` instead of actual newlines. So it looks like your shell is not expanding them as expected (hence no output when you call that file). Can you please clarify how exactly you are invoking the `prerelease.sh` file and in which shell environment.

pirobot, 2016-09-26:
Yes, I copied the output from the terminal exactly.  I see the `> wrapper_scripts/git.py` in the file `prerelease_build_underlay.sh` itself but it does not appear on the terminal (I don't think it should, right?).  Besides, the `wrapper_scripts/git.py` files is getting created.

I'm using bash as my shell and I'm running the `prerelease.sh script` either like this:

`./prerelease.sh`

or like this:

`sh ./prerelease.sh`

Either way gives the same result.

dirk-thomas, 2016-09-26:
Your shell is not expanding the escaped characters (`\n`) into actual newlinew. Therefore the wrapper script is basically empty (except the shebang line).

I don't know what the difference in your shell is compared to the "standard". I was only able to reproduce the same problem when changing the shebang line from `sh` to `bash`. You might want to try the modified `ros_buildfarm` package from the referenced PR. It performs the expansion differently (using `printf` instead of `echo`).

pirobot, 2016-09-26:
Yes, the PR appears to fix it.  I am using bash version 4.3-7ubuntu1.5 and my .bashrc file looks like the following:

```
# ~/.bashrc: executed by bash(1) for non-login shells.
# see /usr/share/doc/bash/examples/startup-files (in the package bash-doc)
# for examples

# If not running interactively, don't do anything
case $- in
    *i*) ;;
      *) return;;
esac

# don't put duplicate lines or lines starting with space in the history.
# See bash(1) for more options
HISTCONTROL=ignoreboth

# append to the history file, don't overwrite it
shopt -s histappend

# for setting history length see HISTSIZE and HISTFILESIZE in bash(1)
HISTSIZE=1000
HISTFILESIZE=2000

# check the window size after each command and, if necessary,
# update the values of LINES and COLUMNS.
shopt -s checkwinsize

# If set, the pattern "**" used in a pathname expansion context will
# match all files and zero or more directories and subdirectories.
#shopt -s globstar

# make less more friendly for non-text input files, see lesspipe(1)
[ -x /usr/bin/lesspipe ] && eval "$(SHELL=/bin/sh lesspipe)"

# set variable identifying the chroot you work in (used in the prompt below)
if [ -z "${debian_chroot:-}" ] && [ -r /etc/debian_chroot ]; then
    debian_chroot=$(cat /etc/debian_chroot)
fi

# set a fancy prompt (non-color, unless we know we "want" color)
case "$TERM" in
    xterm-color) color_prompt=yes;;
esac

# uncomment for a colored prompt, if the terminal has the capability; turned
# off by default to not distract the user: the focus in a terminal window
# should be on the output of commands, not on the prompt
#force_color_prompt=yes

if [ -n "$force_color_prompt" ]; then
    if [ -x /usr/bin/tput ] && tput setaf 1 >&/dev/null; then
	# We have color support; assume it's compliant with Ecma-48
	# (ISO/IEC-6429). (Lack of such support is extremely rare, and such
	# a case would tend to support setf rather than setaf.)
	color_prompt=yes
    else
	color_prompt=
    fi
fi

if [ "$color_prompt" = yes ]; then
    PS1='${debian_chroot:+($debian_chroot)}\[\033[01;32m\]\u@\h\[\033[00m\]:\[\033[01;34m\]\w\[\033[00m\]\$ '
else
    PS1='${debian_chroot:+($debian_chroot)}\u@\h:[\W]\$ '
fi
unset color_prompt force_color_prompt

# If this is an xterm set the title to user@host:dir
case "$TERM" in
xterm*|rxvt*)
    PS1="\[\e]0;${debian_chroot:+($debian_chroot)}\u@\h: \w\a\]$PS1"
    ;;
*)
    ;;
esac

# enable color support of ls and also add handy aliases
if [ -x /usr/bin/dircolors ]; then
    test -r ~/.dircolors && eval "$(dircolors -b ~/.dircolors)" || eval "$(dircolors -b)"
    alias ls='ls --color=auto'
    #alias dir='dir --color=auto'
    #alias vdir='vdir --color=auto'

    alias grep='grep --color=auto'
    alias fgrep='fgrep --color=auto'
    alias egrep='egrep --color=auto'
fi

# some more ls aliases
alias ll='ls -alF'
alias la='ls -A'
alias l='ls -CF'
alias trash='\rm -f *~'
alias cp='cp -i'
alias mv='mv -i'
alias rm='rm -i'
alias iv='rosrun image_view image_view image:=/camera/rgb/image_raw'

# Set environment for local ROS
function remote_ros () {
    eval `~/bin/remoteros "$@"`
}
alias lr='eval `~/bin/localros`'
alias mr='eval `~/bin/localros`'
alias rr="~/bin/remoteros"
alias nr='eval `~/bin/nullros`'

# Add an "alert" alias for long running commands.  Use like so:
#   sleep 10; alert
alias alert='notify-send --urgency=low -i "$([ $? = 0 ] && echo terminal || echo error)" "$(history|tail -n1|sed -e '\''s/^\s*[0-9]\+\s*//;s/[;&|]\s*alert$//'\'')"'

# Alias definitions.
# You may want to put all your additions into a separate file like
# ~/.bash_aliases, instead of adding them here directly.
# See /usr/share/doc/bash-doc/examples in the bash-doc package.

if [ -f ~/.bash_aliases ]; then
    . ~/.bash_aliases
fi

# enable programmable completion features (you don't need to enable
# this, if it's already enabled in /etc/bash.bashrc and /etc/profile
# sources /etc/bash.bashrc).
if ! shopt -oq posix; then
  if [ -f /usr/share/bash-completion/bash_completion ]; then
    . /usr/share/bash-completion/bash_completion
  elif [ -f /etc/bash_completion ]; then
    . /etc/bash_completion
  fi
fi
#source /opt/ros/indigo/setup.bash
source ~/catkin_ws/devel/setup.bash

export ROS_HOSTNAME=`hostname`.local
export ROS_MASTER_URI=http://`hostname`.local:11311
```

dirk-thomas, 2016-09-26:
It is more likely related to your `sh` version (rather then `bash`). I will close this ticket then, thank for providing feedback. A new version of the `ros_buildfarm` package will be available in the near future.

closed 2016-09-26 by dirk-thomas

pirobot, 2016-09-26:
Thanks for your help!