ros/rosdistro #12894: rosdep init fails to download file under Ubuntu 16.04 Arm64

marting87, 2016-10-05:
`sudo rosdep init` fails under Ubuntu 16.04 Arm64. Tested on my Nvidia TX1. 

rosdep version: `python-rosdep/xenial,now 0.11.5-1 all`

rosdep is unable to download over https because of a certificate problem. 

I've bypassed the certificate check by adding the following line to: `/usr/lib/python2.7/dist-packages/rosdep2/main.py`, just after the imports:

`import ssl
ssl._create_default_https_context = ssl._create_unverified_context`

It works, but it does not seem safe..

wstool has a similar issue. Downloading source packages fails because of a certificate error:

 `wstool init -j8 src kinetic-desktop-full-wet.rosinstall`

gavanderhoorn, 2016-10-05:
So, what was the exact error?

marting87, 2016-10-05:
_ERROR: cannot download default sources list from:
https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
Website may be down._

Firefox is loading the website correctly on the same device.

gavanderhoorn, 2016-10-05:
Is this a problem on our end, or a github issue?

marting87, 2016-10-05:
I can confirm it affects all NVIDIA TX1 boards running Ubuntu 16.04 (JetPack 2.3).

gavanderhoorn, 2016-10-05:
Yes, I believe you :). But what I was wondering is if this is a misconfiguration of github, or an actual issue with the way `rosdep` deals with the certificates.

Given the fact that you've spent some time working around it, I'm guessing you've also read up on the problem, and possible causes. Hence my question.

dirk-thomas, 2016-10-05:
Have you tried to install `ca-certificates` (see https://github.com/ros-infrastructure/rosdep/pull/454/files)?

marting87, 2016-10-05:
I'm not sure if it's a github issue or a rosdep issue.. The python script calls `urlopen` to open a https destination. 

It might be a system-wide issue specific to the TX1 as well. If i call wget:

`wget https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list`

, i get the following:

_--2016-10-05 15:14:46--  https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/sources.list.d/20-default.list
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.60.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.60.133|:443... connected.
ERROR: cannot verify raw.githubusercontent.com's certificate, issued by 'CN=DigiCert SHA2 High Assurance Server CA,OU=www.digicert.com,O=DigiCert Inc,C=US':
  Unable to locally verify the issuer's authority.
To connect to raw.githubusercontent.com insecurely, use `--no-check-certificate'._

With the _--no-check-certificate_ flag wget is successful.

gavanderhoorn, 2016-10-05:
@dirk-thomas: that was going to be my next question/comment :)

marting87, 2016-10-05:
@dirk-thomas, `ca-certificates` is already installed.

gavanderhoorn, 2016-10-05:
What is the output of `openssl s_client -connect https://raw.githubusercontent.com` (from [here](http://www.cyberciti.biz/tips/debugging-ssl-communications-from-unix-shell-prompt.html))?

gavanderhoorn, 2016-10-05:
Doing that locally on an Ubuntu 12.04 machine gets me:
```
[..]
depth=1 C = US, O = DigiCert Inc, OU = www.digicert.com, CN = DigiCert SHA2 High Assurance Server CA
verify error:num=20:unable to get local issuer certificate
verify return:0
[..]
```

Which seems to be essentially the same error as @marting87's `wget` is reporting.

According to [this](http://stackoverflow.com/a/23351633) SO answer, that means the client is missing a root certificate. Interestingly, I can still use `wget` to fetch the file from github.

marting87, 2016-10-05:
CONNECTED(00000003)
depth=1 C = US, O = DigiCert Inc, OU = www.digicert.com, CN = DigiCert SHA2 High Assurance Server CA
verify error:num=20:unable to get local issuer certificate

Certificate chain
 0 s:/C=US/ST=California/L=San Francisco/O=Fastly, Inc./CN=www.github.com
   i:/C=US/O=DigiCert Inc/OU=www.digicert.com/CN=DigiCert SHA2 High Assurance Server CA
 1 s:/C=US/O=DigiCert Inc/OU=www.digicert.com/CN=DigiCert SHA2 High Assurance Server CA
   i:/C=US/O=DigiCert Inc/OU=www.digicert.com/CN=DigiCert High Assurance EV Root CA

Server certificate
**[certificate]**

subject=/C=US/ST=California/L=San Francisco/O=Fastly, Inc./CN=www.github.com
issuer=/C=US/O=DigiCert Inc/OU=www.digicert.com/CN=DigiCert SHA2 High Assurance Server CA

No client certificate CA names sent
Peer signing digest: SHA512
Server Temp Key: ECDH, P-256, 256 bits

SSL handshake has read 3326 bytes and written 431 bytes

New, TLSv1/SSLv3, Cipher is ECDHE-RSA-AES128-GCM-SHA256
Server public key is 2048 bit
Secure Renegotiation IS supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
SSL-Session:
    Protocol  : TLSv1.2
    Cipher    : ECDHE-RSA-AES128-GCM-SHA256
    Session-ID: 9684727C280C5D61F7AE31621235D9D8FAF625D9DBEF03B95D16CE201656E32A
    Session-ID-ctx: 
    Master-Key: 4D840603C48440B81CC2DC5BCB8C2C1A5EB52EDF06574E5F6879886941400E36A818D2635578DD2B6507DDE7B2204C21
    Key-Arg   : None
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    TLS session ticket lifetime hint: 1200 (seconds)
    TLS session ticket:
    **[ticket]**
    Start Time: 1475681910
    Timeout   : 300 (sec)
    Verify return code: 20 (unable to get local issuer certificate)

gavanderhoorn, 2016-10-05:
Please enclose the `s_client` output in triple backticks (`).

marting87, 2016-10-05:
@gavanderhoorn might it be possible that `wget` in 12.04 does not verify the ssl certificate? As a security update `wget` might be stricter in 16.04.

tfoote, 2016-10-05:
There's something odd about how the certificates on the TX1 image were installed. I found [here](https://devtalk.nvidia.com/default/topic/952041/ros-install-on-jetson-tx1-e-unable-to-locate-package-ros-indigo-desktop-full-my-thoughts/?offset=19) that the solution is to call 

`sudo c_rehash /etc/ssl/certs`

After calling that rosdep init and wget both work fine.

marting87, 2016-10-07:
Thanks @tfoote! It seems like a TX1 issue rather than a rosdep/github issue. I'll close this issue for now.

closed 2016-10-07 by marting87