ros/rosdistro #12971: Rule for zbar missing libffi?

chrisspen, 2016-10-19:
I'm trying to make a package that uses the libzbar-cffi PyPI package, which requires the Ubuntu packages `libffi-dev`, `libffi6`, `libzbar-dev`. I noticed `libzbar-dev` is included in your `zbar` rule. Should these other packages be added, or should a new rule be created?

Also, if I want rosdep to install `libzbar-cffi` via pip, how would I do that? There is no system package for it, nor is it likely there will ever be.

tfoote, 2016-10-19:
Please see the [documentation for contributing rosdep rules](http://docs.ros.org/independent/api/rosdep/html/contributing_rules.html) 

There is support for pip installations, but it cannot be used as a dependency of a package being released. 

See the rosdep yaml format: http://docs.ros.org/independent/api/rosdep/html/rosdep_yaml_format.html#rosdep-yaml

I believe that the pip installer [does support a depends field](https://github.com/ros-infrastructure/rosdep/blob/master/src/rosdep2/platforms/pip.py#L101). It's not part of the general specification but [there is an allowance for it](http://www.ros.org/reps/rep-0111.html#dependencies) to support cross package manager dependencies. It's not commonly used but there are a few examples in the [homebrew file](https://github.com/ros/rosdistro/blob/master/rosdep/osx-homebrew.yaml#L664-L668)

tfoote, 2017-01-25:
Closing this as there appears to be no action. A PR adding the rosdep rule you need would be appreciated.

closed 2017-01-25 by tfoote