ros/rosdistro #13104: MacOS Sierra - Missing dependency pycurl for urlgrabber

spaghetti-, 2016-10-29:
homebrew python, I'll track it down later but putting it up here so that I don't forget.

```❯ rosdep install --from-paths src --ignore-src --rosdistro kinetic -y
executing command [sudo -H pip install -U urlgrabber]
Collecting urlgrabber
  Downloading urlgrabber-3.9.1.tar.gz (72kB)
    100% |████████████████████████████████| 81kB 867kB/s
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/private/tmp/pip-build-zScnSC/urlgrabber/setup.py", line 3, in <module>
        import urlgrabber as _urlgrabber
      File "urlgrabber/__init__.py", line 54, in <module>
        from grabber import urlgrab, urlopen, urlread
      File "urlgrabber/grabber.py", line 427, in <module>
        import pycurl
    ImportError: No module named pycurl

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /private/tmp/pip-build-zScnSC/urlgrabber/
ERROR: the following rosdeps failed to install
  pip: command [sudo -H pip install -U urlgrabber] failed
```

tfoote, 2016-11-28:
It looks like urlgrabber is missing an install_requires key for pycurl

From: http://urlgrabber.baseurl.org/ it says to file ticket here: and I find this one: https://bugzilla.redhat.com/show_bug.cgi?id=1057150 which appears to cover it/is linked to https://bugzilla.redhat.com/show_bug.cgi?id=1200091

The rpm packages have the dependency inserted correctly: https://www.rpmfind.net/linux/RPM/centos/6.8/i386/Packages/python-urlgrabber-3.9.1-11.el6.noarch.html

The tickets appear to go to the packaging team. Maybe pinging the mailing list about the inability to use it from pip would be worth it. https://lists.baseurl.org/mailman/listinfo/yum-devel

spaghetti-, 2016-11-28:
According to http://yum.baseurl.org/gitweb?p=urlgrabber.git;a=commitdiff;h=e879aa8b7dd4f2f47ef6941ba6381a0eeafb5a13 pycurl is not actually required for `urlgrabber` to function. Patch applied upstream seems to be for v4.

RPM packages seem to have the dependency inserted correctly (well, as correct as it can be in this case) for legacy support.

tfoote, 2017-08-14:
@spaghetti- quick reminder that this was here. I'm going to close it as we're not planning to track this specific dependency overall.

closed 2017-08-14 by tfoote

spaghetti-, 2017-08-22:
@tfoote No problem - I didn't get around to resolving it but on my latest install I didn't run into this error. I think its safe to close it as fixed.