ros/rosdistro #13298: MacOS Sierra - Incompatible OpenCV Version in Kinetic

igilitschenski, 2016-11-25:
When installing kinetic on MacOS Sierra, the compilation of opencv3 fails with the error `'QTKit/QTKit.h' file not found` (as Apple has removed this in the new macOS). This seems to be known (see e.g. homebrew/homebrew-science#4104, opencv/opencv#6913) and resolved by opencv/opencv#7159 which is merged in opencv master. However, this fix seems to not have made it into ROS yet. What would be the best way to incorporate it?

tfoote, 2016-11-26:
Please make a request on the packaging repo for opencv3: https://github.com/ros-gbp/opencv3-release/issues to make a new release.

tfoote, 2016-11-28:
Closing as duplicate of https://github.com/ros-gbp/opencv3-release/issues/6

closed 2016-11-28 by tfoote