ros/rosdistro #13333: Installation indigo on OSX failed while building 'nodelet_tutorial_math'

ejalaa12, 2016-12-01:
Hello,

I followed the official installation of ROS Indigo on OSX.  
Everything worked perfectly until I started building the catkin Workspace.  

And **nodelet_tutorial_math** failed to build but not for the same reason that is answered in the installation tutorial. This is what I get:

```bash
> cd /Users/ejalaa/Development/ros_installation/my_install/build_isolated/nodelet_tutorial_math && /Users/ejalaa/Development/ros_installation/my_install/install_isolated/env.sh make -j4 -l4

[ 50%] Building CXX object CMakeFiles/nodelet_math.dir/src/plus.cpp.o
In file included from /Users/ejalaa/Development/ros_installation/my_install/src/common_tutorials/nodelet_tutorial_math/src/plus.cpp:30:
In file included from /Users/ejalaa/Development/ros_installation/my_install/install_isolated/include/pluginlib/class_list_macros.h:40:
In file included from /Users/ejalaa/Development/ros_installation/my_install/install_isolated/include/class_loader/class_loader.h:33:
In file included from /usr/local/include/boost/shared_ptr.hpp:17:
In file included from /usr/local/include/boost/smart_ptr/shared_ptr.hpp:23:
In file included from /usr/local/include/boost/config/no_tr1/memory.hpp:21:
In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:610:
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cstring:79:9: error:
      no member named 'strcoll' in the global namespace
using ::strcoll;
      ~~^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cstring:80:9: error:
      no member named 'strxfrm' in the global namespace
using ::strxfrm;
      ~~^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cstring:82:9: error:
      no member named 'memchr' in the global namespace; did you mean 'wmemchr'?
using ::memchr;
      ~~^
/usr/include/wchar.h:150:10: note: 'wmemchr' declared here
wchar_t *wmemchr(const wchar_t *, wchar_t, size_t);
         ^
In file included from /Users/ejalaa/Development/ros_installation/my_install/src/common_tutorials/nodelet_tutorial_math/src/plus.cpp:30:
In file included from /Users/ejalaa/Development/ros_installation/my_install/install_isolated/include/pluginlib/class_list_macros.h:40:
In file included from /Users/ejalaa/Development/ros_installation/my_install/install_isolated/include/class_loader/class_loader.h:33:
In file included from /usr/local/include/boost/shared_ptr.hpp:17:
In file included from /usr/local/include/boost/smart_ptr/shared_ptr.hpp:23:
In file included from /usr/local/include/boost/config/no_tr1/memory.hpp:21:
In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:610:
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cstring:86:9: error:
      no member named 'strcspn' in the global namespace
using ::strcspn;
      ~~^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cstring:88:9: error:
      no member named 'strpbrk' in the global namespace
using ::strpbrk;
      ~~^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cstring:90:9: error:
      no member named 'strrchr' in the global namespace; did you mean 'strchr'?
using ::strrchr;
      ~~^
/System/Library/Frameworks/Kernel.framework/Headers/string.h:81:14: note:
      'strchr' declared here
extern char     *strchr(const char *s, int c);
                 ^
In file included from /Users/ejalaa/Development/ros_installation/my_install/src/common_tutorials/nodelet_tutorial_math/src/plus.cpp:30:
In file included from /Users/ejalaa/Development/ros_installation/my_install/install_isolated/include/pluginlib/class_list_macros.h:40:
In file included from /Users/ejalaa/Development/ros_installation/my_install/install_isolated/include/class_loader/class_loader.h:33:
In file included from /usr/local/include/boost/shared_ptr.hpp:17:
In file included from /usr/local/include/boost/smart_ptr/shared_ptr.hpp:23:
In file included from /usr/local/include/boost/config/no_tr1/memory.hpp:21:
In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/memory:610:
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cstring:92:9: error:
      no member named 'strspn' in the global namespace
using ::strspn;
      ~~^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cstring:94:9: error:
      no member named 'strstr' in the global namespace; did you mean 'strnstr'?
using ::strstr;
[... 106 lines cut ...]
In file included from /usr/local/include/boost/system/system_error.hpp:14:
In file included from /usr/local/include/boost/system/error_code.hpp:19:
In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/ostream:140:
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/locale:871:44: error:
      use of undeclared identifier 'errno'
        typename remove_reference<decltype(errno)>::type __save_errno = errno;
                                           ^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/locale:871:53: error:
      no type named 'type' in the global namespace
        typename remove_reference<decltype(errno)>::type __save_errno = errno;
                                                  ~~^
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/locale:871:73: error:
      use of undeclared identifier 'errno'
        typename remove_reference<decltype(errno)>::type __save_errno = errno;
                                                                        ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[2]: *** [CMakeFiles/nodelet_math.dir/src/plus.cpp.o] Error 1
make[1]: *** [CMakeFiles/nodelet_math.dir/all] Error 2
make: *** [all] Error 2
```

I can't figure out what to do ?

Thank you

spaghetti-, 2016-12-28:
This can be solved by passing `-DCMAKE_FIND_FRAMEWORK=LAST` as the very first CMake argument.

wjwwood, 2017-01-03:
I don't know how to better fix this (better than the work around). If anyone know how please speak up. I might look into it again next time I update OS X.

spaghetti-, 2017-01-04:
AFAIK adding the CMake argument to the toplevel cmake list (with an if clause) does not introduce any regressions.

qingxia0702, 2017-01-12:
hi,ejalaa12 
    I have confuse with the same promble,Did you resolve it？

qingxia0702, 2017-01-13:
Sorry for bad English, yesterday I had update macOS to Sierra and use Xcode 8.2.1 to compile it ,But the problem is still here. I have no idea now, anyone know it?

jundazhu, 2017-01-23:
The problem is due to the inclusion of the wrong `string.h`(from kernel framework instead of the standard library).  `-DCMAKE_FIND_FRAMEWORK=LAST` did not resolve the problem on my side. 

I found the culprit to be the FindUUID CMake module. It is located at ${ros_ws}/src/cmake_modules/cmake/Modules/FindUUID.cmake. Changing from find_path(UUID_INCLUDE_DIRS uuid/uuid.h) to find_path(UUID_INCLUDE_DIRS ossp/uuid.h) worked for me. Be sure to install `ossp-uuid `via homebrew first.

bgromov, 2017-01-24:
@jundazhu Thanks for discovering it.

Note for others: to make this solution work it might be necessary to remove `build_isolated/nodelet_core` and `devel_isolated/nodelet_core` directories first (did the trick for me).

argrento, 2017-03-07:
@jundazhu, your solution does not work for me, even with the @bgromov's note.

Also adding `-DCMAKE_FIND_FRAMEWORK=LAST` to the build workspace command `./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST` do not work.

bgromov, 2017-03-08:
@argrento Have you tried to wipe the whole `build_isolated` and `devel_isolated`? That what worked for me on another macOS.

argrento, 2017-03-08:
@bgromov yes, thus I exclude 'nodelet_tutorial_math' from the build tree.

mikaelarguedas, 2017-03-08:
@argrento can you try to downgrade class_loader to [version 3.4](https://github.com/ros/class_loader/commit/d7a3dcb6b29803b5a7ef4118108a31244cc709ed) to identify if this is the root of the problem? (you can post your comments on https://github.com/ros/class_loader/issues/44)  
Thanks!

argrento, 2017-03-09:
@mikaelarguedas yesterday I tried to use [Mac OS X installer](https://github.com/mikepurvis/ros-install-osx). I added `-DCMAKE_FIND_FRAMEWORK=LAST` to [the build options](https://github.com/mikepurvis/ros-install-osx/blob/master/install#L156) and I saw no error, all was good.

mikaelarguedas, 2017-03-09:
@argrento glad it worked for you too. So you did succeed to compile `nodelet_tutorial_math` using @jundazhu's solution? Or is it still excluded from your workspace?

folkwang, 2017-03-14:
@argrento what macOS version are you using? I'm on Sierra and the installer script you mentioned will fail in qt discovery for me.

argrento, 2017-03-14:
@folkwang I have 10.11.6. I have an issue with Qt as well. [Section 4.2](http://wiki.ros.org/kinetic/Installation/OSX/Homebrew/Source#Troubleshooting) helped me.

folkwang, 2017-03-14:
@argrento ah yeah. i did a clean rebuild and it works. thanks!

ilidar, 2017-06-20:
Had same problem with macOS Sierra 10.15.2 and ROS Kinetic, all suggestions didn't work for me, so I've removed `${UUID_INCLUDE_DIRS}` `${UUID_LIBRARIES}` from nodelet package.

Before:
```
catkin_package(
  INCLUDE_DIRS include ${UUID_INCLUDE_DIRS}
  LIBRARIES nodeletlib ${UUID_LIBRARIES}
  CATKIN_DEPENDS bondcpp message_runtime pluginlib rosconsole roscpp std_msgs
  DEPENDS Boost
)
```
After:
```
catkin_package(
  INCLUDE_DIRS include
  LIBRARIES nodeletlib
  CATKIN_DEPENDS bondcpp message_runtime pluginlib rosconsole roscpp std_msgs
  DEPENDS Boost
)
```
Hopefully this could help someone and won't brake something else :P.

Joschka1, 2017-06-28:
@ilidar thank you, that fixed it for me to and now it compiles fine :) 
it also fixed the rqt_gui_cpp error for me!

tfoote, 2017-08-14:
Since this has a workaround could someone please add this to the http://wiki.ros.org/kinetic/Installation/OSX/Homebrew/Source and then we can close this ticket?

ejalaa12, 2017-08-19:
Hi, sorry everyone, I haven't followed this issue closely lately. Besides, I don't get this error anymore, probably because I did several updates meanwhile. However, It seems that this issue has been resolved. Should I close the ticket or wait for someone to add it to the tutorial ? @tfoote

tfoote, 2017-08-19:
If this is no longer an issue with recent updates we can close this out as we don't need to update the documentation. Thanks for the followup @ejalaa12

closed 2017-08-19 by tfoote

Tobias-Fischer, 2018-04-26:
This problem still seems to happen with `lunar`. It would be great if someone could add this to the Homebrew troubleshooting page.